enazoe/yolo-tensorrt #1: detect speed

guods, 2020-01-17:
Thank you for your work, but there are some questions. The speed of the three inference precison (INT8 FP16 FP32) for the one model is the same,  I only modify the config.inference_precison?

enazoe, 2020-01-19:
@guods Hi，What is your platform？PC or nano？windows or ubuntu？And the INT8 precision need the calibration images.

guods, 2020-01-19:
1. environment：PC， Win10， GeForce GTX TITan X， TensorRT 5.1.5.0, CUDA 10.0, CUDNN7.5.0
2. I run the INT8 model with the calibration images,   but the speed of the three inference precison (INT8 FP16 FP32) for the one model is the same.

enazoe, 2020-01-19:
@guods I just test the inference time on my platform(WIN10,GTX1050)，the inference time is 200ms(FP32) and 60ms(INT8).
How long is the calculation time on your platform？
-  check the program is generate the engine file and table file? 
-  delete the engine and table files,re-build the engine

guods, 2020-01-19:
@enazoe 
yolov3-416x416 | gtx1050 | INT8 | 25ms
yolov3-416x416 | gtx1050 | FLOAT32 | 50ms
the result on Ubuntu?
INT8: the inference time on my platform is 17ms, the time of detection one image is 23ms.

enazoe, 2020-01-19:
@guods no ,the time in readme.md is 416*416 model, 200ms(FP32) and 60ms(INT8)is 608*608 , the detect function include the cv::resize method and other image preprocessing.And the opencv resize() method is slow at first call . In my mind, the tian x is more power to get low differnece detect time . And you can use for loop test one image more times.

guods, 2020-01-19:
1. The time of detection one image is 23ms:
get the time by calculating the mean time of the 1000 times on the same image, and it dose not include the time of reading image. 
2. I delete the engine and table files, re-build the engine. I get the same time on INT8 and FP32; I did not find else questions by debugging the code.

enazoe, 2020-01-19:
oh，try yolov3 608x608 model ?，I get a different time of int8 and fp32.

enazoe, 2020-01-19:
@guods do not average the detect time and print per loop time

guods, 2020-01-19:
I try yolov3 608x608 model，I get a  different time of int8 and fp32 (difference 2ms2~3ms) on TiTan X;  on 940mx: difference 10ms

guods, 2020-01-19:
Thanks for your reply, I will try else GPU and PC.

closed 2020-01-19 by guods