enazoe/yolo-tensorrt #2: About the detetction precision

guods, 2020-01-20:
@enazoe Did you evaluate the detection precision? Too much accuracy reduction compared to no acceleration.

enazoe, 2020-01-20:
@guods Yes，I evaluate it my own simple dataset . It does not have significant performance reduction. But for the INT8 precision ,the calibration images best to cover all dataset . In other word, keep the sample diversity.

guods, 2020-01-20:
@enazoe In order to avoid the inferences of calibration images,  I detected my dataset on FP16，the detection results are also poor .

junhua-zhang, 2020-10-23:
@guods @enazoe Correct cv4 act for bottleneck_csp to leaky relu, and precision problem should be solved.

guods, 2020-10-23:
@junhua-zhang thanks, I would like to test it.

closed 2020-10-23 by guods