enazoe/yolo-tensorrt #5: 尺寸不是1：1

zhangxiaopang88, 2020-03-17:
您好，您试过尺寸不是1：1的模型吗，我试尺寸是1：1的测试成功，不是1：1的测试结果为空

enazoe, 2020-03-17:
@zhangxiaopang88 没有测试过，如果模型输入尺寸不是1：1，那么模型里对应的kernel步长之类的可能也要跟着改，但是非标准的可能trt不支持，你可以多测试下

zhangxiaopang88, 2020-03-17:
我把关于长宽不相等的断言给注掉了，       // assert(inputDims.d[0] == inputDims.d[1]);
        assert(kernelSize.d[0] == kernelSize.d[1]);
        assert(stride.d[0] == stride.d[1]);
        assert(padding.d[0] == padding.d[1]); 其他的条件都满足，您看还需要修改哪些地方，能给个建议吗

enazoe, 2020-03-17:
@enazoe 个人感觉是trt不支持这种不规则的filters

zhangxiaopang88, 2020-03-17:
我看到输入尺寸不一致的例子，应该是支持的输入尺寸不一致的情况，我再查查资料 @enazoe

xuhao1, 2020-03-23:
@zhangxiaopang88  I am also dealing with input size inequality, have you find any solution?
(My Linux Notebook has no pinyin input method...)

xuhao1, 2020-03-23:
If simply comment all assert, the network gives a lot of wrong detection results...

zhangxiaopang88, 2020-03-26:
我尝试再次对比，我改动的并不大，只是涉及GridSize的地方改成宽高，处理1：1输入的话不报异常，但是不是1：1的话就会有问题，我准备把yolo转成onnx试试能不能成功

aybora, 2020-05-12:
Hey @zhangxiaopang88 and @xuhao1, I also have inequal sized input images. Have you any progress on that issue?

enazoe, 2020-08-27:
reference this commit 2edde41546bd58f13f2f51c4c9ecbe966f3267a8