enazoe/yolo-tensorrt #6: batchsize 只能设置为1吗

zhangxiaopang88, 2020-03-17:
您好，我想修改测试的batchsize，只是修改_p_net = std::unique_ptr<Yolo>{ new YoloV3(1, _yolo_info, _infer_param) };这里好像不行，请问还有别的地方需要修改吗 @enazoe

enazoe, 2020-03-17:
[https://github.com/enazoe/yolo-tensorrt/blob/5ffe5869c50868c24193d317fa0be6b9d8f8e995/modules/class_yolo_detector.hpp#L84](url)这里和它相关的地方

enazoe, 2020-03-17:
https://github.com/enazoe/yolo-tensorrt/blob/5ffe5869c50868c24193d317fa0be6b9d8f8e995/modules/class_yolo_detector.hpp#L84   
这里和相关的地方

zhangxiaopang88, 2020-03-26:
好像增大batchsize会提高速度，我准备试试，但是还是有一些问题 @enazoe

chongzhong, 2020-08-04:
我试了下batchsize=2，yolov3可以正常推理，而yolov4只能得到第一张图片的结果，第二张结果为空。耗时为batchsize=1的两倍左右

jstumpin, 2020-08-06:
**yolov4-kHALF.engine** (inputSize= 320x320, batchSize = 2) is working fine as depicted below:
[image]

**yolov3-tiny-kHALF.engine** (inputSize= 608x608, batchSize = 2) is also working fine as depicted below:
[image]

But **yolov4-tiny-kHALF.engine** (inputSize= 608x608, batchSize = 2) is working fine for the first image (batchId = 0) and not for the second one (batchId = 1):
[image]

enazoe, 2020-08-06:
@jstumpin @zhangxiaopang88 @chongzhong sorry, batch inference not support yet ,I will working on it and support soon. And welcom PR at same time.

jstumpin, 2020-08-07:
Everything's fixed now, kudos @enazoe for the prompt action! **yolov4-tiny-kHALF-batch2.engine**:
[image]

**yolov4-tiny-kHALF-batch2.engine** (using [_person.jpg_](https://github.com/enazoe/yolo-tensorrt/blob/master/configs/person.jpg) + [_dog.jpg_](https://github.com/enazoe/yolo-tensorrt/blob/master/configs/dog.jpg) example as per in [_result.jpg_](https://github.com/enazoe/yolo-tensorrt/blob/master/configs/result.jpg); slight difference due to yolov4 vs. yolov4-tiny):
[image]

**yolov4-tiny-kHALF-batch4.engine** (add `std::string batchSize;` after [yolo.h#L56](https://github.com/enazoe/yolo-tensorrt/blob/master/modules/yolo.h#L56), add `_yolo_info.batchSize = "batch" + std::to_string(_config.n_max_batch);` after [class_yolo_detector.hpp#L138](https://github.com/enazoe/yolo-tensorrt/blob/master/modules/class_yolo_detector.hpp#L138), change `_yolo_info.enginePath = dataPath + "-" + _yolo_info.precision + "-" + _yolo_info.batchSize + ".engine";` to [class_yolo_detector.hpp#L144](https://github.com/enazoe/yolo-tensorrt/blob/master/modules/class_yolo_detector.hpp#L144), and add `config_v4_tiny.n_max_batch = 4;` after [sample_detector.cpp#L37](https://github.com/enazoe/yolo-tensorrt/blob/master/samples/sample_detector.cpp#L37)):
[image]

enazoe, 2020-08-07:
@jstumpin good idea，and  welcome pr  :)