enazoe/yolo-tensorrt #8: 你好，请问一下，用int8精度的模式怎样使用？

121649982, 2020-04-01:
需要自己加上些图像吗？

enazoe, 2020-04-01:
@121649982 
`config.inference_precison = INT8;
config.calibration_image_list_file_txt = "../configs/calibration_images.txt";`
int8校准的时候需要一些图像，用calibration_image_list_file_txt指定图片的路径

121649982, 2020-04-01:
> @121649982
> `config.inference_precison = INT8; config.calibration_image_list_file_txt = "../configs/calibration_images.txt";`
> int8校准的时候需要一些图像，用calibration_image_list_file_txt指定图片的路径

请问一下，如果用自己训练的模型，这些图像可以是用于训练的吗？感谢

enazoe, 2020-04-01:
@121649982 就是需要你训练darknet模型时候的图片作为trt校准的图片，不需要全部，一个子集就行，几百张

121649982, 2020-04-01:
@enazoe 好的，谢谢了

enazoe, 2020-04-01:
@121649982 注意这个子集要是训练集的均匀采样，如果觉得还不错点个star。。。。

closed 2020-04-01 by enazoe

121649982, 2020-04-01:
好的，感谢，你的代码真的很赞，我在windows上跑起来了

Stars-Chan, 2021-04-26:
在Linux上注意把calibration_images.txt文件里面的”\“换成”/“

henbucuoshanghai, 2021-06-02:
慢着    为什么需要deepstream   ？   训练的代码在哪里呢  yolo345？

wangxudong-cq, 2022-04-15:
> arknet模型时候的图片作为trt校准的
这个图片集是自动下载的？
还是需要自己去找？

wangxudong-cq, 2022-04-15:
> @121649982 `config.inference_precison = INT8; config.calibration_image_list_file_txt = "../configs/calibration_images.txt";` int8校准的时候需要一些图像，用calibration_image_list_file_txt指定图片的路径

这个数据集是自动下载的吗？

enazoe, 2022-04-15:
> > @121649982 `config.inference_precison = INT8; config.calibration_image_list_file_txt = "../configs/calibration_images.txt";` int8校准的时候需要一些图像，用calibration_image_list_file_txt指定图片的路径
> 
> 这个数据集是自动下载的吗？
https://github.com/enazoe/yolo-tensorrt/issues/8#issuecomment-607185879 
https://github.com/enazoe/yolo-tensorrt/releases/tag/calibration_images.txt

wangxudong-cq, 2022-04-15:
@enazoe thank you!
I have copied the images from the Coco training dataset。