enazoe/yolo-tensorrt #12: Build error

anisghaoui, 2020-05-07:
Hi,

I was building on Ubuntu 18.04 with OpenCV installed.

The cmake went fine. Then, I started to make and I got this error : 

```make
make
Scanning dependencies of target detector
[  7%] Building CXX object CMakeFiles/detector.dir/modules/calibrator.cpp.o
In file included from /home/anis/TensorRT/yolo-tensorrt/modules/ds_image.h:28:0,
                 from /home/anis/TensorRT/yolo-tensorrt/modules/calibrator.h:29,
                 from /home/anis/TensorRT/yolo-tensorrt/modules/calibrator.cpp:26:
/home/anis/TensorRT/yolo-tensorrt/modules/trt_utils.h:34:10: fatal error: opencv2/dnn/dnn.hpp: No such file or directory
 #include <opencv2/dnn/dnn.hpp>
          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
CMakeFiles/detector.dir/build.make:243: recipe for target 'CMakeFiles/detector.dir/modules/calibrator.cpp.o' failed
make[2]: *** [CMakeFiles/detector.dir/modules/calibrator.cpp.o] Error 1
CMakeFiles/Makefile2:124: recipe for target 'CMakeFiles/detector.dir/all' failed
make[1]: *** [CMakeFiles/detector.dir/all] Error 2
Makefile:103: recipe for target 'all' failed
make: *** [all] Error 2
```

I rechecked the openCV build and I couldn't find anything missing.

if anyone has feedback on this, please hit me.

enazoe, 2020-05-08:
@anisghaoui hi，the dnn moudle is in the opencv contrib modules, make sure your opencv build with the contrib module . And ,make sure the cmake could find the opencv.pc or opencv.cmake file.

closed 2020-05-11 by enazoe

anisghaoui, 2020-05-11:
Hi again, I checked the build  of opencv and ensured that I positioned the all DNN modules in the contrib modules with the correct path.  Your Cmake can indeed find the correct Opencv version it is only the opencvdnn header that's somehow missing. 
I honesty don't understand why it is missing while the build show it has been done correctly

Thank you.

anisghaoui, 2020-05-11:
I might be missing a flag. This the cmake config :

```cmakeDetected processor: x86_64
Looking for ccache - not found
libjpeg-turbo: VERSION = 2.0.4, BUILD = opencv-4.3.0-dev-libjpeg-turbo
Could NOT find OpenJPEG (minimal suitable version: 2.0, recommended version >= 2.3.1)
Found OpenEXR: /usr/lib/x86_64-linux-gnu/libIlmImf.so
Checking for module 'gtk+-3.0'
  No package 'gtk+-3.0' found
found Intel IPP (ICV version): 2020.0.0 [2020.0.0 Gold]
at: /home/anis/opencv/build/3rdparty/ippicv/ippicv_lnx/icv
Cannot find Intel IPP Integration Wrappers, optimizations will be limited. Use IPPIWROOT to set custom location
CUDA detected: 10.2
CUDA NVCC target flags: -gencode;arch=compute_30,code=sm_30;-gencode;arch=compute_35,code=sm_35;-gencode;arch=compute_37,code=sm_37;-gencode;arch=compute_50,code=sm_50;-gencode;arch=compute_52,code=sm_52;-gencode;arch=compute_60,code=sm_60;-gencode;arch=compute_61,code=sm_61;-gencode;arch=compute_70,code=sm_70;-gencode;arch=compute_75,code=sm_75;-D_FORCE_INLINES
Could not find OpenBLAS include. Turning OpenBLAS_FOUND off
Could not find OpenBLAS lib. Turning OpenBLAS_FOUND off
Could NOT find Atlas (missing: Atlas_CBLAS_INCLUDE_DIR Atlas_CLAPACK_INCLUDE_DIR Atlas_CBLAS_LIBRARY Atlas_BLAS_LIBRARY Atlas_LAPACK_LIBRARY) 
A library with BLAS API not found. Please specify library location.
LAPACK requires BLAS
A library with LAPACK API not found. Please specify library location.
VTK is not found. Please set -DVTK_DIR in CMake to VTK build directory, or to VTK install subdirectory with VTKConfig.cmake file
OpenCV Python: during development append to PYTHONPATH: /home/anis/opencv/build/python_loader
Checking for module 'gstreamer-base-1.0'
  No package 'gstreamer-base-1.0' found
Checking for module 'gstreamer-app-1.0'
  No package 'gstreamer-app-1.0' found
Checking for module 'gstreamer-riff-1.0'
  No package 'gstreamer-riff-1.0' found
Checking for module 'gstreamer-pbutils-1.0'
  No package 'gstreamer-pbutils-1.0' found
Module opencv_alphamat disabled because the following dependencies are not found: Eigen
Caffe:   NO
Protobuf:   NO
Glog:   NO
freetype2:   YES (ver 21.0.15)
harfbuzz:    YES (ver 1.7.2)
Could NOT find HDF5 (missing: HDF5_LIBRARIES HDF5_INCLUDE_DIRS) (found version "")
Module opencv_ovis disabled because OGRE3D was not found
No preference for use of exported gflags CMake configuration set, and no hints for include/library directories provided. Defaulting to preferring an installed/exported gflags CMake configuration if available.
Found installed version of gflags: /usr/lib/x86_64-linux-gnu/cmake/gflags
Detected gflags version: 2.2.1
Failed to find glog - Could not find glog include directory, set GLOG_INCLUDE_DIR to directory containing glog/logging.h
Module opencv_sfm disabled because the following dependencies are not found: Eigen Glog/Gflags
Checking for module 'tesseract'
  No package 'tesseract' found
Tesseract:   NO
Processing WORLD modules...
    module opencv_cudev...
    module opencv_core...
Allocator metrics storage type: 'long long'
    module opencv_cudaarithm...
    module opencv_flann...
    module opencv_imgproc...
    module opencv_intensity_transform...
    module opencv_ml...
    module opencv_phase_unwrapping...
    module opencv_plot...
    module opencv_quality...
    module opencv_reg...
    module opencv_surface_matching...
    module opencv_cudafilters...
    module opencv_cudaimgproc...
    module opencv_cudawarping...
    module opencv_dnn...
Registering hook 'INIT_MODULE_SOURCES_opencv_dnn': /home/anis/opencv/modules/dnn/cmake/hooks/INIT_MODULE_SOURCES_opencv_dnn.cmake
    module opencv_dnn_superres...
    module opencv_features2d...
    module opencv_freetype...
    module opencv_fuzzy...
    module opencv_hfs...
[... 130 lines cut ...]
           at:                   /home/anis/opencv/build/3rdparty/ippicv/ippicv_lnx/icv
    Intel IPP IW:                NO
    Lapack:                      NO
    Eigen:                       NO
    OpenVX:                      NO
    Custom HAL:                  NO
    Protobuf:                    build (3.5.1)

  NVIDIA CUDA:                   YES (ver 10.2, CUFFT CUBLAS)
    NVIDIA GPU arch:             30 35 37 50 52 60 61 70 75
    NVIDIA PTX archs:

  cuDNN:                         YES (ver 7.6.5)

  OpenCL:                        YES (no extra features)
    Include path:                /home/anis/opencv/3rdparty/include/opencl/1.2
    Link libraries:              Dynamic load

  Python (for build):            /usr/bin/python2.7

  Install to:                    /usr/local
-----------------------------------------------------------------

Configuring done
```

anisghaoui, 2020-05-12:
Hi again,

I have found what was missing. In your Cmake you placed absolute paths to dependencies. So I replaced the line them with the one for Linux and somehow missed one.

I changed : `target_include_directories(detector PRIVATE modules/ D:/dependency/gflag/include ${CUDA_TOOLKIT_ROOT_DIR}/include D:/dependency/TensorRT-5.1.5.0_cuda-10.0_cudnn-7.5/include D:/dependency/OPENCV341/include extra/ ${CMAKE_CUDA_TOOLKIT_INCLUDE_DIRECTORIES})`
  
to 

`target_include_directories(detector PRIVATE modules/ ${CUDA_TOOLKIT_ROOT_DIR}/include extra/ ${CMAKE_CUDA_TOOLKIT_INCLUDE_DIRECTORIES} ${OpenCV_INCLUDE_DIRS})
`

This make and compiles correctly.

commit c57ed8b8 (2020-05-21): fix #12 #14