enazoe/yolo-tensorrt #14: fatal error: opencv2/opencv.hpp: No such file or directory

stiv-yakovenko, 2020-05-17:
I have installed opencv on my jetson nano. But I get this error:

```
stiv@nano:~/yolo-tensorrt/build$ cmake ..
-- gflags found
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found CUDA: /usr/local/cuda (found version "10.0") 
-- CUDA version:$(CUDA_VERSION)
-- CUDA 10 detected
-- Found OpenCV: /usr/local (found version "4.1.2") 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/stiv/yolo-tensorrt/build
stiv@nano:~/yolo-tensorrt/build$ make
[  7%] Building NVCC (Device) object CMakeFiles/detector.dir/modules/detector_generated_kernel.cu.o
Scanning dependencies of target detector
[ 14%] Building CXX object CMakeFiles/detector.dir/modules/calibrator.cpp.o
In file included from /home/stiv/yolo-tensorrt/modules/ds_image.h:28:0,
                 from /home/stiv/yolo-tensorrt/modules/calibrator.h:29,
                 from /home/stiv/yolo-tensorrt/modules/calibrator.cpp:26:
/home/stiv/yolo-tensorrt/modules/trt_utils.h:32:10: fatal error: opencv2/opencv.hpp: No such file or directory
 #include <opencv2/opencv.hpp>
          ^~~~~~~~~~~~~~~~~~~~
compilation terminated.
CMakeFiles/detector.dir/build.make:69: recipe for target 'CMakeFiles/detector.dir/modules/calibrator.cpp.o' failed
make[2]: *** [CMakeFiles/detector.dir/modules/calibrator.cpp.o] Error 1
CMakeFiles/Makefile2:104: recipe for target 'CMakeFiles/detector.dir/all' failed
make[1]: *** [CMakeFiles/detector.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
```

You can see that I do have opencv in place:

```
stiv@nano:~/yolo-tensorrt/build$ locate opencv2/opencv.hpp
/home/stiv/jsoft/opencv-4.1.2/include/opencv2/opencv.hpp
/home/stiv/jsoft-1.0/opencv-4.1.2/include/opencv2/opencv.hpp
/usr/include/opencv4/opencv2/opencv.hpp
/usr/local/include/opencv4/opencv2/opencv.hpp
```

can we fix it somehow?

anisghaoui, 2020-05-18:
I already encountered this problem here #12 .

enazoe, 2020-05-18:
@stiv-yakovenko @anisghaoui sorry I'm busy on other work these days and I will have a test on my computer. Thx!

enazoe, 2020-05-21:
@anisghaoui @stiv-yakovenko add the opencv include dir to the detector.so include path
```
target_include_directories(detector PRIVATE modules/ ${OpenCV_INCLUDE_DIRS} ${CUDA_TOOLKIT_ROOT_DIR}/include ${CMAKE_CUDA_TOOLKIT_INCLUDE_DIRECTORIES})
```

closed 2020-07-17 by enazoe