enazoe/yolo-tensorrt #18: undefined symbol

npumazehong, 2020-07-14:
undefined symbol: _ZN8Detector6detectERKN2cv3MatERSt6vectorI6ResultSaIS5_EE
undefined symbol: _ZN8Detector4initERK6Config	

I could run your sample yesterday.But today when I ran this sample,the errors above happened.I try to solve it,but it always show these errors.I know that the undefined symbol happens in "detector_->init(config)" and "detector_->detect()".My device is Nvidia Xavier NX,and the os is Ubuntu 18.04 customed by Nvidia.

enazoe, 2020-07-14:
@npumazehong ok，sorry，I will fix it soon

npumazehong, 2020-07-16:
> @npumazehong ok，sorry，I will fix it soon

So，have you find the error of dynamic link absence? I could not sovle this problem myself. Could you help me fix it soon? Thanks

enazoe, 2020-07-16:
@npumazehong I 'm so sorry, my nx was taken by a colleague，and only wait for him to return to check what went wrong. you could try comment this [line ](https://github.com/enazoe/yolo-tensorrt/blob/master/CMakeLists.txt#L26)and rebuild.

npumazehong, 2020-07-16:
I'm glad to receive your letter.You are such a nice man. I have commentted that line,but it didn't work. I guess there is something wrong happens in linking dynamic libraries or the definition of Detector(YoloDetector) Class's method.   I'm looking forward to your reply.

enazoe, 2020-07-16:
@npumazehong give your wechat account :)

npumazehong, 2020-07-16:
ma770634926

closed 2020-07-21 by enazoe