enazoe/yolo-tensorrt #22: Input width and height

Nuzhny007, 2020-07-17:
Some questions.
1. In yolo.cpp you set condition: assert(m_InputW == m_InputH);
Why? AlexeyAB write: **set network size width=416 height=416 or any value multiple of 32**

2. And second question about this assert:
```
if (weights.size() != weightPtr)
     {
         std::cout << "Number of unused weights left : " << weights.size() - weightPtr << std::endl;
         assert(0);
     }
```
In my yolov4-tiny with size 608x608 this condition goes to assert but in release in works. And **weights.size() < weightPtr**

enazoe, 2020-07-20:
1.yeah , current version does not support daynamic input dimension.
2.the yolov4-tiny is not yet supported, but I am working on it and support soon. Because the yolov4-tiny route layer is different from yolov4. you could check this [link](https://github.com/Tianxiaomo/pytorch-YOLOv4/issues/125). We need add a plugin layer at tensorrt like pytorch chunk to implementation the tiny route layer.

Nuzhny007, 2020-07-20:
Thnx!

closed 2020-07-20 by Nuzhny007