enazoe/yolo-tensorrt #24: Does this project supports for tiny-yolov3?

WangShuo-DaHeng, 2020-07-20:
just like the title.

enazoe, 2020-07-20:
yes
https://github.com/enazoe/yolo-tensorrt/blob/master/modules/class_detector.h#L20

WangShuo-DaHeng, 2020-07-20:
I found NVIDIA-AI-IOT's project has lots of libraries like "tiny-yolov3.h",but in your repo you removed it for some reason.so it seemed that you merged them into one library.By the way, If i want to run this repo with tiny-yolov3 weigths and cfg that trained by myself,which only have 4 classes,do i need to adjust some code?

您好，我查看了nvidia原装工程的代码，发现他们对不同的网络整了很多不同的.c.h，一看您这里没有就来问一句，抱歉占用您点时间。我也是比较小白，我跑通了tiny yolov3，python版本的trt加速，想转换成cpp的但是奈何太菜，python版本的如果是自己训练的网络需要根据class改输出部分的reshape代码，这个cpp版本的需要做什么改动吗？我是tiny yolo 288x288，class数量为4

enazoe, 2020-07-20:
@WangShuo-DaHeng 这个版本只支持darknet的cfg文件，你的模型如果是darknet训练出来的话就没问题，不用做任何修改，但是如果是其他框架训练的话改动就比较大了。

WangShuo-DaHeng, 2020-07-20:
没有问题，就是darknet的，我自己研究一下，谢谢您啦，另外还有个小bug，cmakelist里面的编译器c++11在我电脑上make会报错，得改成c++14。最后我这边拉了一个国内的nvidia jetson developer群，大概有570多人，您要是方便，有空，非常欢迎您来群里交流指点，群号：725916205。

closed 2020-07-20 by WangShuo-DaHeng

enazoe, 2020-07-20:
@WangShuo-DaHeng 好的，谢谢，我发现这个问题了，下个版本会改正。