enazoe/yolo-tensorrt #26: 在jetson tx2 上编译出错，求助

jcyhcs, 2020-07-21:
[image]
[image]

jcyhcs, 2020-07-21:
我看了一下错误，应该是TensorRT版本的问题，因为jetpack4.2 是TensorRT5.0.6,里面根本没有IPluginV2IOExt这个类，请问，能够提供一个基于TensorRT 5 等通用版本的yolov4实现呢？TensorRT 7 太新了，还有很多老板卡比如TX2在被使用。

enazoe, 2020-07-21:
@jcyhcs 不好意思，通知邮件早就收到了，但是今天我这github一直有问题，登不上去，现在用手机回你吧。
这个因为之前适配trt7所以改了好多，v4也基本是用7写的，改回去可能工作量比较大，不行你就回滚到之前的版本，然后再把v4的代码搬过来，应该改动不大，就那几个类换成以前的就行。如果觉得工作量大，把板卡刷机吧。。。。

closed 2020-07-21 by enazoe

TUT-jiayou, 2021-04-16:
@jcyhcs HI, 你好, 我想问一下, 您那边最后在TX2 上部署起来了么? 如果能够成功部署的话,方便告知一下是如何部署起来的么?