enazoe/yolo-tensorrt #27: yolov3-tiny

JiaojiaoHan, 2020-07-21:
Config config_v3;
config_v3.net_type = YOLOV3_TINY;
config_v3.file_model_cfg = "../configs/yolov3-tiny.cfg";
config_v3.file_model_weights = "../configs/yolov3-tiny.weights";
config_v3.calibration_image_list_file_txt = "../configs/calibration_images.txt";
config_v3.inference_precison = INT8;

yolov3, yolov4 and yolov4-tiny run successfully.
but yolov3-tiny failed.

Loading pre-trained weights...
Loading complete!
      layer               inp_size            out_size       weightPtr
(1)   conv-bn-leaky     3 x 416 x 416      16 x 416 x 416    496
(2)   maxpool          16 x 416 x 416      16 x 208 x 208    496
(3)   conv-bn-leaky    16 x 208 x 208      32 x 208 x 208    5232
(4)   maxpool          32 x 208 x 208      32 x 104 x 104    5232
(5)   conv-bn-leaky    32 x 104 x 104      64 x 104 x 104    23920
(6)   maxpool          64 x 104 x 104      64 x  52 x  52    23920
(7)   conv-bn-leaky    64 x  52 x  52     128 x  52 x  52    98160
(8)   maxpool         128 x  52 x  52     128 x  26 x  26    98160
(9)   conv-bn-leaky   128 x  26 x  26     256 x  26 x  26    394096
(10)  maxpool         256 x  26 x  26     256 x  13 x  13    394096
(11)  conv-bn-leaky   256 x  13 x  13     512 x  13 x  13    1575792
(12)  maxpool         512 x  13 x  13     512 x  12 x  12    1575792
(13)  conv-bn-leaky   512 x  12 x  12    1024 x  12 x  12    6298480
(14)  conv-bn-leaky  1024 x  12 x  12     256 x  12 x  12    6561648
(15)  conv-bn-leaky   256 x  12 x  12     512 x  12 x  12    7743344
(16)  conv-linear     512 x  12 x  12     255 x  12 x  12    7874159
(17)  yolo            255 x  12 x  12     255 x  12 x  12    7874159
(18)  route                  -            256 x  12 x  12    7874159
(19)  conv-bn-leaky   256 x  12 x  12     128 x  12 x  12    7907439
(20)  upsample        128 x  12 x  12     128 x  24 x  24        -
ERROR: Assertion failed: d.nbDims >= 1, file f:\image-process-lib\code\tensorrt-7.0.0.11\samples\yolo-tensorrt\modules\trt_utils.cpp, line 409

enazoe, 2020-07-21:
@JiaojiaoHan is a bug，i will fix it soon.

beizhengren, 2020-07-21:
> yolov3, yolov4 and yolov4-tiny run successfully.   
@enazoe 
@JiaojiaoHan 
Hi, there are some errors when I compling the repo.
Have you encountered that ? How do you resolve it? 
Thank you!
```
/home/nvidia/Documents/wyz/darknetTensorRT/yolo-tensorrt/modules/chunk.h(35): error: cannot override "final" function "nvinfer1::IPluginV2IOExt::supportsFormat"
/usr/include/aarch64-linux-gnu/NvInferRuntimeCommon.h(762): here

/home/nvidia/Documents/wyz/darknetTensorRT/yolo-tensorrt/modules/chunk.h(46): warning: function "nvinfer1::IPluginV2Ext::configurePlugin(const nvinfer1::Dims *, int, const nvinfer1::Dims *, int, const nvinfer1::DataType *, const nvinfer1::DataType *, const __nv_bool *, const __nv_bool *, nvinfer1::PluginFormat, int)" is hidden by "nvinfer1::Chunk::configurePlugin" -- virtual function override intended?

/home/nvidia/Documents/wyz/darknetTensorRT/yolo-tensorrt/modules/chunk.cu(70): error: identifier "rsize_t" is undefined

/home/nvidia/Documents/wyz/darknetTensorRT/yolo-tensorrt/modules/chunk.cu(70): error: declaration is incompatible with "size_t nvinfer1::Chunk::getSerializationSize() const"
/home/nvidia/Documents/wyz/darknetTensorRT/yolo-tensorrt/modules/chunk.h(33): here

3 errors detected in the compilation of "/tmp/tmpxft_000045c1_00000000-6_chunk.cpp1.ii".
CMake Error at detector_generated_chunk.cu.o.cmake:280 (message):
  Error generating file
  /home/nvidia/Documents/wyz/darknetTensorRT/yolo-tensorrt/build-nx/CMakeFiles/detector.dir/modules/./detector_generated_chunk.cu.o

CMakeFiles/detector.dir/build.make:334: recipe for target 'CMakeFiles/detector.dir/modules/detector_generated_chunk.cu.o' failed
make[2]: *** [CMakeFiles/detector.dir/modules/detector_generated_chunk.cu.o] Error 1
CMakeFiles/Makefile2:124: recipe for target 'CMakeFiles/detector.dir/all' failed
make[1]: *** [CMakeFiles/detector.dir/all] Error 2
Makefile:103: recipe for target 'all' failed
make: *** [all] Error 2

```

enazoe, 2020-07-21:
@beizhengren which version of jetpack?

beizhengren, 2020-07-22:
@enazoe Hi, 
I used jetson xavier NX with tensorrt 7.1.0-1+cuda10.2   
I have found some information by 
```sh
 cat /etc/nv_tegra_release
```
```# R32 (release), REVISION: 4.2, GCID: 20074772, BOARD: t186ref, EABI: aarch64, DATE: Thu Apr  9 01:26:40 UTC 2020```
And I can complie successfully with dev-yolov4 branch which not support yolov4-tiny.

enazoe, 2020-07-22:
@beizhengren you could switch to master, it support yolov4-tiny now

beizhengren, 2020-07-22:
>  you could switch to master, it support yolov4-tiny now

@enazoe Hi,
I just  used master branch, and encountered the errors https://github.com/enazoe/yolo-tensorrt/issues/27#issuecomment-661752953.
I mean, master can not complie successfully, but dev-yolov4 can.

enazoe, 2020-07-22:
@beizhengren check the repository to the newest bc9446bb05f676084f208767febdc21d2c03157a

enazoe, 2020-07-22:
@beizhengren i will fix it soon

beizhengren, 2020-07-22:
@enazoe Hi
I've resolved it, and run successfully!
Thanks!

enazoe, 2020-07-22:
> Config config_v3;
> config_v3.net_type = YOLOV3_TINY;
> config_v3.file_model_cfg = "../configs/yolov3-tiny.cfg";
> config_v3.file_model_weights = "../configs/yolov3-tiny.weights"; [...]

closed 2020-07-22 by enazoe

enazoe, 2020-07-22:
@Jessicahjj try it 7898f6538a0a002a117465d62af74434d8f5e925

Jessicahjj, 2020-07-23:
@enazoe  Thanks!

myboyhood, 2023-02-08:
Yes! I also meet this problem, and solve it. 
1. I checkout `master` 
2. recompile with `cmake .. && make`
3. delete the old `yolov3-tiny-kun-kHALF-batch1.engine` file, because this file is generate according to the machine and GPU type. Let us to generate a new one.
4.  The successful ouput is as following:
```
darknet_weights_path: /home/zph/vins_ws/src/pnp_target_node/config/yolov3-tiny-kun.weights
 darknet_cfg: /home/zph/vins_ws/src/pnp_target_node/config/yolov3-tiny-kun.cfgFile does not exist : /home/zph/vins_ws/src/pnp_target_node/config/yolov3-tiny-kun-kHALF-batch1.engine
Loading pre-trained weights...
Loading complete!
      layer               inp_size            out_size       weightPtr
(1)   conv-bn-leaky     1 x 416 x 416      16 x 416 x 416    208   
(2)   maxpool          16 x 416 x 416      16 x 208 x 208    208   
(3)   conv-bn-leaky    16 x 208 x 208      32 x 208 x 208    4944  
(4)   maxpool          32 x 208 x 208      32 x 104 x 104    4944  
(5)   conv-bn-leaky    32 x 104 x 104      64 x 104 x 104    23632 
(6)   maxpool          64 x 104 x 104      64 x  52 x  52    23632 
(7)   conv-bn-leaky    64 x  52 x  52     128 x  52 x  52    97872 
(8)   maxpool         128 x  52 x  52     128 x  26 x  26    97872 
(9)   conv-bn-leaky   128 x  26 x  26     256 x  26 x  26    393808
(10)  maxpool         256 x  26 x  26     256 x  13 x  13    393808
(11)  conv-bn-leaky   256 x  13 x  13     512 x  13 x  13    1575504
(12)  maxpool         512 x  13 x  13     512 x  13 x  13    1575504
(13)  conv-bn-leaky   512 x  13 x  13    1024 x  13 x  13    6298192
(14)  conv-bn-leaky  1024 x  13 x  13     256 x  13 x  13    6561360
(15)  conv-bn-leaky   256 x  13 x  13     512 x  13 x  13    7743056
(16)  conv-linear     512 x  13 x  13      18 x  13 x  13    7752290
(17)  yolo             18 x  13 x  13      18 x  13 x  13    7752290
(18)  route                  -            256 x  13 x  13    7752290
(19)  conv-bn-leaky   256 x  13 x  13     128 x  13 x  13    7785570
(20)  upsample        128 x  13 x  13     128 x  26 x  26        - 
(21)  route                  -            384 x  26 x  26    7785570
(22)  conv-bn-leaky   384 x  26 x  26     256 x  26 x  26    8671330
(23)  conv-linear     256 x  26 x  26      18 x  26 x  26    8675956
(24)  yolo             18 x  26 x  26      18 x  26 x  26    8675956
File does not exist : /home/zph/vins_ws/src/pnp_target_node/config/yolov3-tiny-kun-kHALF-batch1.engine
Building the TensorRT Engine...
Building complete!
Serializing the TensorRT Engine...
Serialized plan file cached at location : /home/zph/vins_ws/src/pnp_target_node/config/yolov3-tiny-kun-kHALF-batch1.engine
Loading TRT Engine...
Loading Complete!
TensorRTDetector initialized
```