enazoe/yolo-tensorrt #31: Don't work yolo v4

Nuzhny007, 2020-07-22:
Hi!
After latest update yolov4-tiny works fine but yolov4 failed:

```
Loading pre-trained weights...
Loading complete!
      layer               inp_size            out_size       weightPtr
(1)   conv-bn-mish      3 x 608 x 608      32 x 608 x 608    992   
(2)   conv-bn-mish     32 x 608 x 608      64 x 304 x 304    19680 
(3)   conv-bn-mish     64 x 304 x 304      64 x 304 x 304    24032 
(4)   route                  -             64 x 304 x 304    24032 
(5)   conv-bn-mish     64 x 304 x 304      64 x 304 x 304    28384 
(6)   conv-bn-mish     64 x 304 x 304      32 x 304 x 304    30560 
(7)   conv-bn-mish     32 x 304 x 304      64 x 304 x 304    49248 
(8)   skip             64 x 304 x 304      64 x 304 x 304        - 
(9)   conv-bn-mish     64 x 304 x 304      64 x 304 x 304    53600 
(10)  route                  -            128 x 304 x 304    53600 
(11)  conv-bn-mish    128 x 304 x 304      64 x 304 x 304    62048 
(12)  conv-bn-mish     64 x 304 x 304     128 x 152 x 152    136288
(13)  conv-bn-mish    128 x 152 x 152      64 x 152 x 152    144736
(14)  route                  -            128 x 152 x 152    144736
(15)  conv-bn-mish    128 x 152 x 152      64 x 152 x 152    153184
(16)  conv-bn-mish     64 x 152 x 152      64 x 152 x 152    157536
(17)  conv-bn-mish     64 x 152 x 152      64 x 152 x 152    194656
(18)  skip             64 x 152 x 152      64 x 152 x 152        - 
(19)  conv-bn-mish     64 x 152 x 152      64 x 152 x 152    199008
(20)  conv-bn-mish     64 x 152 x 152      64 x 152 x 152    236128
(21)  skip             64 x 152 x 152      64 x 152 x 152        - 
(22)  conv-bn-mish     64 x 152 x 152      64 x 152 x 152    240480
(23)  route                  -            128 x 152 x 152    240480
(24)  conv-bn-mish    128 x 152 x 152     128 x 152 x 152    257376
(25)  conv-bn-mish    128 x 152 x 152     256 x  76 x  76    553312
(26)  conv-bn-mish    256 x  76 x  76     128 x  76 x  76    586592
(27)  route                  -            256 x  76 x  76    586592
(28)  conv-bn-mish    256 x  76 x  76     128 x  76 x  76    619872
(29)  conv-bn-mish    128 x  76 x  76     128 x  76 x  76    636768
(30)  conv-bn-mish    128 x  76 x  76     128 x  76 x  76    784736
(31)  skip            128 x  76 x  76     128 x  76 x  76        - 
(32)  conv-bn-mish    128 x  76 x  76     128 x  76 x  76    801632
(33)  conv-bn-mish    128 x  76 x  76     128 x  76 x  76    949600
(34)  skip            128 x  76 x  76     128 x  76 x  76        - 
(35)  conv-bn-mish    128 x  76 x  76     128 x  76 x  76    966496
(36)  conv-bn-mish    128 x  76 x  76     128 x  76 x  76    1114464
(37)  skip            128 x  76 x  76     128 x  76 x  76        - 
(38)  conv-bn-mish    128 x  76 x  76     128 x  76 x  76    1131360
(39)  conv-bn-mish    128 x  76 x  76     128 x  76 x  76    1279328
(40)  skip            128 x  76 x  76     128 x  76 x  76        - 
(41)  conv-bn-mish    128 x  76 x  76     128 x  76 x  76    1296224
(42)  conv-bn-mish    128 x  76 x  76     128 x  76 x  76    1444192
(43)  skip            128 x  76 x  76     128 x  76 x  76        - 
(44)  conv-bn-mish    128 x  76 x  76     128 x  76 x  76    1461088
(45)  conv-bn-mish    128 x  76 x  76     128 x  76 x  76    1609056
(46)  skip            128 x  76 x  76     128 x  76 x  76        - 
(47)  conv-bn-mish    128 x  76 x  76     128 x  76 x  76    1625952
(48)  conv-bn-mish    128 x  76 x  76     128 x  76 x  76    1773920
(49)  skip            128 x  76 x  76     128 x  76 x  76        - 
(50)  conv-bn-mish    128 x  76 x  76     128 x  76 x  76    1790816
(51)  conv-bn-mish    128 x  76 x  76     128 x  76 x  76    1938784
(52)  skip            128 x  76 x  76     128 x  76 x  76        - 
(53)  conv-bn-mish    128 x  76 x  76     128 x  76 x  76    1955680
(54)  route                  -            256 x  76 x  76    1955680
(55)  conv-bn-mish    256 x  76 x  76     256 x  76 x  76    2022240
(56)  conv-bn-mish    256 x  76 x  76     512 x  38 x  38    3203936
(57)  conv-bn-mish    512 x  38 x  38     256 x  38 x  38    3336032
(58)  route                  -            512 x  38 x  38    3336032
(59)  conv-bn-mish    512 x  38 x  38     256 x  38 x  38    3468128
(60)  conv-bn-mish    256 x  38 x  38     256 x  38 x  38    3534688
(61)  conv-bn-mish    256 x  38 x  38     256 x  38 x  38    4125536
(62)  skip            256 x  38 x  38     256 x  38 x  38        - 
(63)  conv-bn-mish    256 x  38 x  38     256 x  38 x  38    4192096
(64)  conv-bn-mish    256 x  38 x  38     256 x  38 x  38    4782944
(65)  skip            256 x  38 x  38     256 x  38 x  38        - 
(66)  conv-bn-mish    256 x  38 x  38     256 x  38 x  38    4849504
(67)  conv-bn-mish    256 x  38 x  38     256 x  38 x  38    5440352
(68)  skip            256 x  38 x  38     256 x  38 x  38        - 
(69)  conv-bn-mish    256 x  38 x  38     256 x  38 x  38    5506912
(70)  conv-bn-mish    256 x  38 x  38     256 x  38 x  38    6097760
(71)  skip            256 x  38 x  38     256 x  38 x  38        - 
(72)  conv-bn-mish    256 x  38 x  38     256 x  38 x  38    6164320
(73)  conv-bn-mish    256 x  38 x  38     256 x  38 x  38    6755168
(74)  skip            256 x  38 x  38     256 x  38 x  38        - 
(75)  conv-bn-mish    256 x  38 x  38     256 x  38 x  38    6821728
(76)  conv-bn-mish    256 x  38 x  38     256 x  38 x  38    7412576
(77)  skip            256 x  38 x  38     256 x  38 x  38        - 
(78)  conv-bn-mish    256 x  38 x  38     256 x  38 x  38    7479136
(79)  conv-bn-mish    256 x  38 x  38     256 x  38 x  38    8069984
(80)  skip            256 x  38 x  38     256 x  38 x  38        - 
(81)  conv-bn-mish    256 x  38 x  38     256 x  38 x  38    8136544
(82)  conv-bn-mish    256 x  38 x  38     256 x  38 x  38    8727392
(83)  skip            256 x  38 x  38     256 x  38 x  38        - 
(84)  conv-bn-mish    256 x  38 x  38     256 x  38 x  38    8793952
(85)  route                  -            512 x  38 x  38    8793952
(86)  conv-bn-mish    512 x  38 x  38     512 x  38 x  38    9058144
(87)  conv-bn-mish    512 x  38 x  38    1024 x  19 x  19    13780832
(88)  conv-bn-mish   1024 x  19 x  19     512 x  19 x  19    14307168
(89)  route                  -           1024 x  19 x  19    14307168
(90)  conv-bn-mish   1024 x  19 x  19     512 x  19 x  19    14833504
(91)  conv-bn-mish    512 x  19 x  19     512 x  19 x  19    15097696
(92)  conv-bn-mish    512 x  19 x  19     512 x  19 x  19    17459040
(93)  skip            512 x  19 x  19     512 x  19 x  19        - 
(94)  conv-bn-mish    512 x  19 x  19     512 x  19 x  19    17723232
(95)  conv-bn-mish    512 x  19 x  19     512 x  19 x  19    20084576
(96)  skip            512 x  19 x  19     512 x  19 x  19        - 
(97)  conv-bn-mish    512 x  19 x  19     512 x  19 x  19    20348768
(98)  conv-bn-mish    512 x  19 x  19     512 x  19 x  19    22710112
(99)  skip            512 x  19 x  19     512 x  19 x  19        - 
(100) conv-bn-mish    512 x  19 x  19     512 x  19 x  19    22974304
(101) conv-bn-mish    512 x  19 x  19     512 x  19 x  19    25335648
(102) skip            512 x  19 x  19     512 x  19 x  19        - 
(103) conv-bn-mish    512 x  19 x  19     512 x  19 x  19    25599840
(104) route                  -           1024 x  19 x  19    25599840
(105) conv-bn-mish   1024 x  19 x  19    1024 x  19 x  19    26652512
(106) conv-bn-leaky  1024 x  19 x  19     512 x  19 x  19    27178848
(107) conv-bn-leaky   512 x  19 x  19    1024 x  19 x  19    31901536
(108) conv-bn-leaky  1024 x  19 x  19     512 x  19 x  19    32427872
(109) maxpool         512 x  19 x  19     512 x  15 x  15    32427872
(110) route                  -            512 x  19 x  19    32427872
(111) maxpool         512 x  19 x  19     512 x  11 x  11    32427872
(112) route                  -            512 x  19 x  19    32427872
(113) maxpool         512 x  19 x  19     512 x   7 x   7    32427872
ERROR: ERROR: (114) route                  -           32567               32427872
ERROR: ERROR: ERROR: (115) conv-bn-leaky  32567               32567               33478496
ERROR: ERROR: ERROR: ERROR: (116) conv-bn-leaky  32567               32567               33482592
ERROR: ERROR: ERROR: ERROR: (117) conv-bn-leaky  32567               32567               33484640
ERROR: ERROR: ERROR: ERROR: (118) conv-bn-leaky  32567               32567               33485664
ERROR: ERROR: ERROR: Segmentation fault (core dumped)

```

enazoe, 2020-07-22:
@Nuzhny007 which paltform?

Nuzhny007, 2020-07-22:
Ubuntu 18.04, CUDA 10.0, tensorrt 7.0.0.11

And are yolov4 on your platform good? Then I can check difference between old and New code and find problem

Nuzhny007, 2020-07-22:
Sorry, my bad. All works

closed 2020-07-22 by Nuzhny007

enazoe, 2020-07-23:
Never mind.