enazoe/yolo-tensorrt #33: windows下编译出错

qinxianglinya, 2020-07-23:
博主，你好。请问该项目支持在tensorrt6.0.1.5下编译吗？win10+tensorrt6.0.1.5+cuda10.0+cudnn7.6.5编译出错了。

enazoe, 2020-07-23:
需要回滚到之前的版本，但是以前版本不支持yolov4

qinxianglinya, 2020-07-23:
> 需要回滚到之前的版本，但是以前版本不支持yolov4

qinxianglinya, 2020-07-23:
> > 需要回滚到之前的版本，但是以前版本不支持yolov4

目前我是在编译yolov4，这个tensorrt6.0.1.5不支持吗？

closed 2020-07-23 by qinxianglinya

enazoe, 2020-07-23:
@qinxianglinya 嗯，v4使用的新api写的，弃用了trt即将移除的api，你是必须用trt6？没必要吧，trt7肯定是有些提升的

qinxianglinya, 2020-07-23:
> @qinxianglinya 嗯，v4使用的新api写的，弃用了trt即将移除的api，你是必须用trt6？没必要吧，trt7肯定是有些提升的

因为我之前一个算法是用tensorrt 6.0.1部署的，那个算法是用tensorflow写的，用的uff解析器。我想把两个算法集成到一个框架下，请问需要改哪些地方使他支持tensorrt6呢？

enazoe, 2020-07-23:
@qinxianglinya 
1.我觉你原来的算法如果可以直接解析的话，说明没有什么特殊的层（如果是dl模型的话），那么我觉得trt7也是可以解析你的算法。
2.如果想把我这个改回去的话，你就改报错对应的类就行，新api和旧的名字类似。

qinxianglinya, 2020-07-23:
> @qinxianglinya
> 1.我觉你原来的算法如果可以直接解析的话，说明没有什么特殊的层（如果是dl模型的话），那么我觉得trt7也是可以解析你的算法。
> 2.如果想把我这个改回去的话，你就改报错对应的类就行，新api和旧的名字类似。

好的。谢谢。