enazoe/yolo-tensorrt #38: Error while making

xNeorem, 2020-07-28:
Hello I got this error while making the code.

/usr/bin/ld: cannot find -lgflags
collect2: error: ld returned 1 exit status
CMakeFiles/detector.dir/build.make:1039: recipe for target 'libdetector.so' failed
make[2]: *** [libdetector.so] Error 1
CMakeFiles/Makefile2:104: recipe for target 'CMakeFiles/detector.dir/all' failed
make[1]: *** [CMakeFiles/detector.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

I read that " The project generate the libdetector.so lib, and the sample code. If you want to use the generated libdetector.so lib in your own project,the cmake file perhaps could help you in scripts dir."

I don't know if it is related to my problem... I am a beginner of jetson and deepstream platforms.

Thanks for your help

enazoe, 2020-07-29:
try it 86f3297a92197869e4f3cabaa3a21bd4bfb922ae  ？

xNeorem, 2020-07-29:
it works

closed 2020-07-29 by xNeorem