enazoe/yolo-tensorrt #40: Error while linking libdetector.so

xNeorem, 2020-08-04:
Hello, I  got this error in compiling :

**../../src_file/libdetector.so: undefined reference to `cv::dnn::dnn4_v20190621::blobFromImages(cv::_InputArray const&, double, cv::Size_<int>, cv::Scalar_<double> const&, bool, bool, int)'**

I think that I am linking wrong the library in my own project, how can I link correctly?

I am using a MakeFile not cmake...Can you give me an example of how to import and use the library and use it in your project?

enazoe, 2020-08-04:
@xNeorem sry, I know little about makefile. I think you could reference this cmake file https://github.com/enazoe/yolo-tensorrt/blob/master/scripts/CMakeLists.txt .It is a example of use libdetector.so in your own project through set the lib path.

xNeorem, 2020-08-05:
Thanks, i used your cmakelist and it worked

closed 2020-08-05 by xNeorem