enazoe/yolo-tensorrt #45: Error while building TensorRT Engine

xNeorem, 2020-08-11:
Hello i got this error:

Building the TensorRT Engine...
ERROR: ERROR: yolo-trt: /home/nvidia/Documents/yolo-tensorrt/modules/yolo.cpp:522: void Yolo::createYOLOEngine(nvinfer1::DataType, Int8EntropyCalibrator*): Assertion `m_Engine != nullptr' failed.
Aborted (core dumped)

I am using jetpack 4.3 , CUDA : 10.0.326 , OpencCV : 4.1.2, TensorRT: 6.0.1.10 on nano

Any suggest ??

enazoe, 2020-08-12:
try update the jetpack to 4.4

xNeorem, 2020-08-12:
I cannot do that, because I need a face recognition and in 4.4 is extremely slow...

enazoe, 2020-08-12:
@xNeorem try delete the line **-gencode arch** at cmakelists or set it as _compute_53,code=sm_53_ on nano.

enazoe, 2020-08-12:
#13

xNeorem, 2020-08-12:
I tried but still not working...

beybars1, 2020-08-19:
Hey man, have you managed to resolve this issue? I have an exactly same error on JN system.

yudiprtm, 2021-11-25:
I have same issue, even though using jetpack 4.6. @enazoe any ideas?

Nuzhny007, 2021-11-25:
File with weight not opened or corrupted

yudiprtm, 2021-11-26:
but it's worked for FP16 and FP32