enazoe/yolo-tensorrt #46: 有关前向推断的问题

beizhengren, 2020-08-12:
作者您好,
#### 1.在创建engine的过程
我看您是把所有层都解析完成之后, 才判断是否有已经产生好的engine. 可不可以把这个判断放在最前面?
https://github.com/enazoe/yolo-tensorrt/blob/91455c6bc26e85e2e4f26164e3c1404ab9f39b06/modules/yolo.cpp#L475

#### 2.关于前向推断
我想直接读取保存好的engine 然后进行推断. 也就是直接
```
detector_->detect(mat_temp, res);
``` 
而不进行初始化去创建engine
```
detector_->init(config_v4_tiny);
```
 但是这样做是有错的
```
Process finished with exit code 139 (interrupted by signal 11: SIGSEGV)
```
请问这里我应该如何实现比较好?

感谢!

enazoe, 2020-08-12:
1.可以放到最前面
2.程序会根据你的模型自动去找对应的engine，如果没有才会生成engin，有的话会直接加载。再就是engin文件是和平台绑定的并不是所有平台通用的

beizhengren, 2020-08-12:
@enazoe 
作者您好, 我放到了
https://github.com/enazoe/yolo-tensorrt/blob/91455c6bc26e85e2e4f26164e3c1404ab9f39b06/modules/yolo.cpp#L177
的后面, 即:
```cpp
    trtWeights.push_back(divWeights);

    if (fileExists(m_EnginePath))
    {
        std::cout << "Using previously generated plan file located at " << m_EnginePath
                  << std::endl;
        destroyNetworkUtils(trtWeights);
        return;
    }
```
会报以下错误:
```cpp
ERROR: yolo-trt: ~/yolo-tensorrt/modules/yolo.cpp:578: 
void Yolo::allocateBuffers(): Assertion `(tensor.bindingIndex != -1) && "Invalid output binding index"' failed.
```

beizhengren, 2020-08-17:
@enazoe 
作者您好,我找到原因了,是因为对输出的内存分配和确认问题导致的.
现在解决了,谢谢~

closed 2020-08-17 by beizhengren