enazoe/yolo-tensorrt #47: sln issue (Windows)

Whisper94, 2020-08-18:
hello. I have some problems with sln for windows. Could you help to fix it? I have already changed all custom dependencies but still get some errors and I don't know how to handle with it.

[image]

I have also several warnings (30). Most of them about conversion like "size_t" to "unit32_t" or initialization  like "double" to "const float"

Could you help me to fix my issue? Thank you!

enazoe, 2020-08-18:
@Whisper94 the version of your visual studio?

Whisper94, 2020-08-18:
@enazoe I have 2019. Also 2015/2017/2019 redistributable. Is it not compatible with 2015 which you required?
Thank you for your fast answer!!

enazoe, 2020-08-19:
@Whisper94 try vs2015，，I think the vs2019 compiler removed the support for <experimental/filesystem>

Whisper94, 2020-08-21:
Ok. Thx. At the moment I have no chance to install 2015 version and deinstall 2019.I'll Look another way.

freeluojing123, 2022-03-05:
Add _SILENCE_EXPERIMENTAL_FILESYSTEM_DEPRECATION_WARNING to Preprocessor definitions.

Project -> Properties -> C/C++ -> Preprocessor -> Preprocessor Definitions.