enazoe/yolo-tensorrt #50: 关于cfg,weights,engine文件

lq0104, 2020-09-03:
抱歉还要打扰一下，我还有个小白问题，之前没做过tensorRT。我看到如果没有engine文件，程序就会根据cfg和weights来默认生成一个engine文件，下次启动时如果已有engine文件就可以直接使用，我看到engine文件也足够的大，超过了weights的大小，我的问题是：
engine文件是否已经包含了cfg和weights的所有信息，以便一旦生成engine文件后就不再需要cfg和weights文件了，这样的话模型的整体占用空间就会少一半，因为毕竟yolov3/4/5原始的模型还是很大的；还是说每次启动时cfg和weights是必须的规避不了？我现在看到的现象是如果没有找到cfg和weights，程序就跑飞了

enazoe, 2020-09-03:
@lq0104 呃，你这问题提的太是时候了，最新得commit会检查是否已经存在engine文件，存在则不会再加载weights文件；再就是其实engine只包含了网络里面得结构信息，但是后面nms，解析之类得信息是不包含的，也就是说用到trt api部分的会保存，所以需要加载cfg文件，获取一部分信息来进行post处理

lq0104, 2020-09-03:
太强了！我更新了代码，测试没问题，这个repo是我找到的几个支持yolov4-tensorRT中最适配Windows的代码，谢谢大佬的分享~
还要请教一个问题： #22 这里面问到的第一个问题，他问的应该是如果input width和height不一致的话代码支不支持，我看您的回答好像是如果对网络动态输入width和height不支持，看着有点迷糊
所以在这我还想再确认一下目前的状：
1.如果输入width!=height的话支不支持？比如输入不是（416，416）而是（512，384）
2.如果动态输入分辨率支不支持？比如这次是（416，416），下次是（512，512），再下次是（608，608）。

enazoe, 2020-09-03:
@lq0104 这个我当时理解的有偏差，其实动态输入(每次inference长宽不固定)和模型长宽不想等(32倍数)是两回事。v5是支持长宽不等的，但是不支持动态输入，也就是build完engine就不能改了.现在v3和v4还不支持长宽不想等输入，不过好改，如果你有需要，我明天改下。

lq0104, 2020-09-03:
太感谢了！如果方便的话，有空的时间麻烦改一下就行，我前天刚接触tensorRT，太菜了，要不还能试着PR一下

enazoe, 2020-09-03:
ok

lq0104, 2020-09-04:
谢谢代码更新！
在更新下我这边对engine的测试，我发现如果两台电脑GPU的计算能力不同的话，这台电脑生成的engine文件在另一台电脑上会报错，信息类似于：expecting compute 7.5 got compute 6.1
解决方法是如果计算能力不同，那么还是要把weights文件拷过去，让另一台电脑重新生成engine文件。原因我在网上查的解释是：“tensorrt 的序列化和反序列化操作只能在特定硬件上做，两个操作需配套；”
在这里共享一下，以便有遇到同样问题的人能够快速解决 )

enazoe, 2020-09-04:
@lq0104 是的，你说的很对