enazoe/yolo-tensorrt #52: yolov5-3.0的推理速度

chongzhong, 2020-09-04:
您好，我在jetson nano平台上测试了yolov3-tiny和yolov5的性能

| model       | input size | batchsize | precision | inference time |
| ----------- | ---------- | --------- | --------- | -------------- |
| yolov3-tiny | 416x416    | 2         | FP16      | ~60ms          |
| yolov5s-3.0 | 320x320    | 2         | FP16      | ~120ms         |
| yolov5m-3.0 | 640x320    | 2         | FP16      | ~230ms         |

有个疑问，为什么yolov5s和yolov5m模型比yolov3-tiny小，而推理速度确更慢？

之前我也通过python 测试过yolov5m（pytorch->onnx->tensorrt），而推理耗时在100ms左右，比这个C++的版本快些。

enazoe, 2020-09-04:
@chongzhong 模型并没有比v3-tiny小，模型输入的大小不代表模型大小，v5的两个模型的深度是比v3-tiny大的多，你测试的onnx版本的时间是inference时间还是detect时间（包含pre和post），我这个打印的时间是detect时间，你要想测试inference时间可以把https://github.com/enazoe/yolo-tensorrt/blob/master/modules/yolo.cpp#L922 和936行的注释去掉再测试下

chongzhong, 2020-09-04:
@enazoe 谢谢，onnx的包括了pre和post

enazoe, 2020-09-04:
@chongzhong onnx的版本batch size也是2吗？

chongzhong, 2020-09-04:
是的。我弄错了，yolov5m的flops的确更多。

closed 2020-09-04 by chongzhong

chongzhong, 2020-09-04:
抱歉，python测试的是yolov5s, jetson nano平台，batchsize=2，尺寸320x320，推理速度（包括pre和post）约110ms, 和本项目速度差不多。

enazoe, 2020-09-04:
@chongzhong 哦哦