enazoe/yolo-tensorrt #57: it is failed for Yolov4 transfer to tensorrt! Please Help answer

shiguoliang, 2020-09-19:
Thank you very much for sharing the project， but I have a error transferring to yolov4. It is show mistake "Internal error: could not find any implementation for node (Unnamed Layer* 271) [Deconvolution], try increasing the workspace size with IBuilder:: setMaxWorkspaceSize() ../builder/tacticOptimizer.cpp (1523) - OutOfMemory Error in computeCosts: 0", 
Please help me to answer， thanks very much

enazoe, 2020-09-20:
@shiguoliang which version of your trt ?

shiguoliang, 2020-09-20:
ubuntu16,TensorRT-7.0.0.11 , cuda 9.0 , cudnn 7.6.3 , opencv4.1 ;

enazoe, 2020-09-20:
@shiguoliang use trt7.1 and recommend version

shiguoliang, 2020-09-20:
thanks very much

yutao007, 2020-09-28:
setMaxWorkspaceSize()将这个改成1<<30   试试