enazoe/yolo-tensorrt #58: yolov5s std::invalid_argument

canuysal, 2020-09-21:
I'm trying to load yolov5s as in the samples, but getting this error. Output is something like

    Loading pre-trained weights...
    Loading complete!

     layer               inp_size            out_size             
     (1)   Focus             3 x 320 x 640      32 x 160 x 320          
     (2)   Conv             32 x 160 x 320      64 x  80 x 160          
     (3)   BottleneckCSP    64 x  80 x 160      64 x  80 x 160          
     (4)   Conv             64 x  80 x 160     128 x  40 x  80          
     (5)   BottleneckCSP   128 x  40 x  80     128 x  40 x  80          
     (6)   Conv            128 x  40 x  80     256 x  20 x  40          
     (7)   BottleneckCSP   256 x  20 x  40     256 x  20 x  40          
     (8)   Conv            256 x  20 x  40     512 x  10 x  20          
     (9)   SPP             512 x  10 x  20     512 x  10 x  20          
     terminate called after throwing an instance of 'std::invalid_argument'
     terminate called recursively

And gdb states that it crashes in

    Yolo::create_engine_yolov5() --->  parse_bottleneck_args() ----> std::__cxx11::stoi()

What might be the issue? Can you help?

enazoe, 2020-09-21:
confirm the compiler support the c++14, and which platform you use？

canuysal, 2020-09-22:
Yes im using gcc 7.5 with c++14 flag on, my platform is jetson tx2. have cuda 10.2 installed

canuysal, 2020-09-22:
I tried using yolov3 and it runs without problems.

enazoe, 2020-09-22:
use the yaml2cfg.py trans the pt file to .weights

canuysal, 2020-09-23:
Yeah I used that to generate cfg and weights as in your tutorial.