enazoe/yolo-tensorrt #59: unspecified launch failure

sdimantsd, 2020-09-24:
Hi! Thanks for you'r work!!
I used you'r repo with tiny yolo, and sometimes i get the following error:
`Cuda failure: unspecified launch failure in file /home/ws/projects/yolo-trt/yolo.cpp at line 924`

Im using Jetson Xavier with cuda 10.0
 Thanks!

enazoe, 2020-09-24:
which version of your jetpack and tensorrt？

sdimantsd, 2020-09-24:
```
$ python jetsonInfo.py
NVIDIA Jetson AGX Xavier [16GB]
 L4T 32.3.1 [ JetPack 4.3 ]
   Ubuntu 18.04.3 LTS
   Kernel Version: 4.9.140-tegra
nvdc: nvdcEventThread select failed: Bad file descriptor
 CUDA 10.0.326
   CUDA Architecture: 7.2
 OpenCV version: 3.4.0
   OpenCV Cuda: NO
 CUDNN: 7.6.3.28
 TensorRT: 6.0.1.10
 Vision Works: 1.6.0.500n
 VPI: 0.1.0
```

Thanks

enazoe, 2020-09-24:
make sure have enough hard disk space when generate the engin file. And upgrade to jetpack4.4 ？I am not test it on jetpack4,3.

sdimantsd, 2020-09-24:
OK, i will try on jetpack4.4.
The engine created and it's running. but somtimes it's crush...

sdimantsd, 2020-11-11:
Hi again :-)
Now im running the file with jetpack4.4

$ python jetsonInfo.py
NVIDIA Jetson Nano (Developer Kit Version)
 L4T 32.4.3 [ JetPack 4.4 ]
   Ubuntu 18.04.5 LTS
   Kernel Version: 4.9.140-tegra
 CUDA 10.2.89
   CUDA Architecture: 5.3
 OpenCV version: 3.4.0
   OpenCV Cuda: NO
 CUDNN: 8.0.0.180
 TensorRT: 7.1.3.0
 Vision Works: 1.6.0.501
 VPI: 4.4-b186

Somtimes m getting this error:
ERROR: FAILED_EXECUTION: std::exception
Cuda failure: unspecified launch failure in file /home/ws/projects/yolo-trt/yolo.cpp at line 931

sdimantsd, 2020-11-12:
OK, it's probably  by mistake...
I used the GPU with many networks on the same time.