enazoe/yolo-tensorrt #61: jetson nano上转换yolov4失败，盼解答

jch-wang, 2020-09-26:
望解答，不胜感激。
>>>>以下是我的jetson nano配置：
JetPack4.4；cuDNN 8.0；CUDA 10.2；OpenCV 4.1.1；Visionworks 1.6；python3   3.6.9 ；TensorRT 7.1.3.4

>>>>以下是运行时的输出：
```
wjc@wjc-desktop:~/yolo-tensorrt/build$ ./yolo-trt
File does not exist : ../configs/yolov4-kFLOAT-batch4.engine
Loading pre-trained weights...
Loading complete!
      layer               inp_size            out_size       weightPtr
(1)   conv-bn-mish      3 x 416 x 416      32 x 416 x 416    992   
(2)   conv-bn-mish     32 x 416 x 416      64 x 208 x 208    19680 
(3)   conv-bn-mish     64 x 208 x 208      64 x 208 x 208    24032 
(4)   route                  -             64 x 208 x 208    24032 
(5)   conv-bn-mish     64 x 208 x 208      64 x 208 x 208    28384 
(6)   conv-bn-mish     64 x 208 x 208      32 x 208 x 208    30560 
(7)   conv-bn-mish     32 x 208 x 208      64 x 208 x 208    49248 
(8)   skip             64 x 208 x 208      64 x 208 x 208        - 
(9)   conv-bn-mish     64 x 208 x 208      64 x 208 x 208    53600 
(10)  route                  -            128 x 208 x 208    53600 
(11)  conv-bn-mish    128 x 208 x 208      64 x 208 x 208    62048 
(12)  conv-bn-mish     64 x 208 x 208     128 x 104 x 104    136288
(13)  conv-bn-mish    128 x 104 x 104      64 x 104 x 104    144736
(14)  route                  -            128 x 104 x 104    144736
(15)  conv-bn-mish    128 x 104 x 104      64 x 104 x 104    153184
(16)  conv-bn-mish     64 x 104 x 104      64 x 104 x 104    157536
(17)  conv-bn-mish     64 x 104 x 104      64 x 104 x 104    194656
(18)  skip             64 x 104 x 104      64 x 104 x 104        - 
(19)  conv-bn-mish     64 x 104 x 104      64 x 104 x 104    199008
(20)  conv-bn-mish     64 x 104 x 104      64 x 104 x 104    236128
(21)  skip             64 x 104 x 104      64 x 104 x 104        - 
(22)  conv-bn-mish     64 x 104 x 104      64 x 104 x 104    240480
(23)  route                  -            128 x 104 x 104    240480
(24)  conv-bn-mish    128 x 104 x 104     128 x 104 x 104    257376
(25)  conv-bn-mish    128 x 104 x 104     256 x  52 x  52    553312
(26)  conv-bn-mish    256 x  52 x  52     128 x  52 x  52    586592
(27)  route                  -            256 x  52 x  52    586592
(28)  conv-bn-mish    256 x  52 x  52     128 x  52 x  52    619872
(29)  conv-bn-mish    128 x  52 x  52     128 x  52 x  52    636768
(30)  conv-bn-mish    128 x  52 x  52     128 x  52 x  52    784736
(31)  skip            128 x  52 x  52     128 x  52 x  52        - 
(32)  conv-bn-mish    128 x  52 x  52     128 x  52 x  52    801632
(33)  conv-bn-mish    128 x  52 x  52     128 x  52 x  52    949600
(34)  skip            128 x  52 x  52     128 x  52 x  52        - 
(35)  conv-bn-mish    128 x  52 x  52     128 x  52 x  52    966496
(36)  conv-bn-mish    128 x  52 x  52     128 x  52 x  52    1114464
(37)  skip            128 x  52 x  52     128 x  52 x  52        - 
(38)  conv-bn-mish    128 x  52 x  52     128 x  52 x  52    1131360
(39)  conv-bn-mish    128 x  52 x  52     128 x  52 x  52    1279328
(40)  skip            128 x  52 x  52     128 x  52 x  52        - 
(41)  conv-bn-mish    128 x  52 x  52     128 x  52 x  52    1296224
(42)  conv-bn-mish    128 x  52 x  52     128 x  52 x  52    1444192
(43)  skip            128 x  52 x  52     128 x  52 x  52        - 
(44)  conv-bn-mish    128 x  52 x  52     128 x  52 x  52    1461088
(45)  conv-bn-mish    128 x  52 x  52     128 x  52 x  52    1609056
(46)  skip            128 x  52 x  52     128 x  52 x  52        - 
(47)  conv-bn-mish    128 x  52 x  52     128 x  52 x  52    1625952
(48)  conv-bn-mish    128 x  52 x  52     128 x  52 x  52    1773920
(49)  skip            128 x  52 x  52     128 x  52 x  52        - 
(50)  conv-bn-mish    128 x  52 x  52     128 x  52 x  52    1790816
(51)  conv-bn-mish    128 x  52 x  52     128 x  52 x  52    1938784
(52)  skip            128 x  52 x  52     128 x  52 x  52        - 
(53)  conv-bn-mish    128 x  52 x  52     128 x  52 x  52    1955680
(54)  route                  -            256 x  52 x  52    1955680
(55)  conv-bn-mish    256 x  52 x  52     256 x  52 x  52    2022240
(56)  conv-bn-mish    256 x  52 x  52     512 x  26 x  26    3203936
(57)  conv-bn-mish    512 x  26 x  26     256 x  26 x  26    3336032
(58)  route                  -            512 x  26 x  26    3336032
(59)  conv-bn-mish    512 x  26 x  26     256 x  26 x  26    3468128
[... 97 lines cut ...]
(157) conv-bn-leaky  1024 x  13 x  13     512 x  13 x  13    54196318
(158) conv-bn-leaky   512 x  13 x  13    1024 x  13 x  13    58919006
(159) conv-bn-leaky  1024 x  13 x  13     512 x  13 x  13    59445342
(160) conv-bn-leaky   512 x  13 x  13    1024 x  13 x  13    64168030
(161) conv-linear    1024 x  13 x  13     255 x  13 x  13    64429405
(162) yolo            255 x  13 x  13     255 x  13 x  13    64429405
File does not exist : ../configs/yolov4-kFLOAT-batch4.engine
Building the TensorRT Engine...
Building complete!
Serializing the TensorRT Engine...
Killed
```
>>>>以下是查看相关存储：
```
wjc@wjc-desktop:~$ df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/mmcblk0p1   59G   19G   38G  34% /
none            1.7G     0  1.7G   0% /dev
tmpfs           2.0G  4.0K  2.0G   1% /dev/shm
tmpfs           2.0G   27M  2.0G   2% /run
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           2.0G     0  2.0G   0% /sys/fs/cgroup
tmpfs           396M   12K  396M   1% /run/user/120
tmpfs           396M  136K  396M   1% /run/user/1000
```

jch-wang, 2020-09-28:
> File does not exist : ../configs/yolov4-kFLOAT-batch4.engine
> Building the TensorRT Engine...
> Building complete!
> Serializing the TensorRT Engine...
> Killed

补充：也报过这个错误：
wjc@wjc-desktop:~/yolo-tensorrt/build$ ./yolo-trt
File does not exist : ../configs/yolov4-kFLOAT-batch4.engine
Loading pre-trained weights...
Loading complete!
.......
......
File does not exist : ../configs/yolov4-kFLOAT-batch4.engine
Building the TensorRT Engine...
Building complete!
Serializing the TensorRT Engine...
ERROR: FAILED_ALLOCATION: std::bad_alloc
yolo-trt: /home/wjc/yolo-tensorrt/modules/yolo.cpp:1251: void Yolo::writePlanFileToDisk(): Assertion `m_ModelStream && "Unable to serialize engine"' failed.
Aborted (core dumped)

yutao007, 2020-09-28:
v4不成功，多半是因为内存不够，用nano的换成半精度试试

jch-wang, 2020-09-28:
> v4不成功，多半是因为内存不够，用nano的换成半精度试试

有道理，在哪个文件里改呀？我尝试尝试

yutao007, 2020-09-28:
config_v4.inference_precison = FP32;
改这个。fp16   int8

jch-wang, 2020-09-28:
> config_v4.inference_precison = FP32;
> 改这个。fp16 int8

晓得了，感谢指点，我去尝试。

enazoe, 2020-09-28:
@yutao007 @jch-wang 是的就是内存问题

Lajiaboer, 2022-07-24:
请问这个yolo-tenssort可以部署在ubuntu上吗，是x86的机器+nvidia GPU，不是jetson系列的板子

enazoe, 2022-07-26:
> 请问这个yolo-tenssort可以部署在ubuntu上吗，是x86的机器+nvidia GPU，不是jetson系列的板子
可以