enazoe/yolo-tensorrt #63: TensorFlow-Yolov3 > onnx > trt

MuhammadAsadJaved, 2020-10-10:
Hi, 
    I am trying to convert TensorFlow-Yolov3  .ckpt or .pb weights to  .onnx and then to .trt engine.   and finally, run in Xavier NX for the interface. 

I am able to convert .pb to .onnx  but having problems converting .onnx to .trt engine.  Is there any method to convert .ckpt weights directly to .trt engine?   I am facing the following errors during conversion. 

+Is there any WeChat group? 

[image]

enazoe, 2020-10-10:
NO,we have qq group 1151955802，this project just support the darknet weights. 
And the problem is that you have a custom layer， the parser can not parse the custom layer.

MuhammadAsadJaved, 2020-10-10:
@enazoe  OK. Thank you.  How to handle the custom layers parsing problem? Any method or suggestions?

enazoe, 2020-10-10:
use the tensorrt api,write a plugin layer

MuhammadAsadJaved, 2020-10-10:
OK. Thank you very much.

closed 2020-10-10 by MuhammadAsadJaved