enazoe/yolo-tensorrt #65: trt_utils.obj : error LNK2001: 無法解析的外部符號 "class cv::Mat __cdecl cv::dnn::dnn4_v20200609::blobFromImages

Scottchou7, 2020-10-13:
感謝您的project，小弟在嘗試build "dll_detector" 時，出現以下錯誤，請大佬指教，謝謝。

```
OS: Windows 10
TensorRT: 7.1.3.4
CUDA: 11.0
Cudnn: 8.0.2
OpenCV: 4.1.2
```

```
1>------ 已開始建置: 專案: dll_detector, 組態: Release x64 ------
1>C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Common7\IDE\VC\VCTargets\Microsoft.CppBuild.targets(391,5): warning MSB8028: 中繼目錄 (D:\TensorRT\yolo-tensorrt-master\sln\x64\) 包含從其他專案 (test_dll.vcxproj) 共用的檔案。這可能導致錯誤的清除和重建行為。
1>   正在建立程式庫 D:\TensorRT\yolo-tensorrt-master\sln\detector.lib 和物件 D:\TensorRT\yolo-tensorrt-master\sln\detector.exp
1>trt_utils.obj : error LNK2001: 無法解析的外部符號 "class cv::Mat __cdecl cv::dnn::dnn4_v20200609::blobFromImages(class cv::_InputArray const &,double,class cv::Size_<int>,class cv::Scalar_<double> const &,bool,bool,int)" (?blobFromImages@dnn4_v20200609@dnn@cv@@YA?AVMat@3@AEBV_InputArray@3@NV?$Size_@H@3@AEBV?$Scalar_@N@3@_N3H@Z)
1>D:\TensorRT\yolo-tensorrt-master\sln\detector.dll : fatal error LNK1120: 1 個無法解析的外部符號
1>專案 "dll_detector.vcxproj" 建置完成 -- 失敗。
========== 建置: 0 成功、1 失敗、0 最新、0 略過 ==========
```

enazoe, 2020-10-13:
使用编译了opencv_contrib的opencv试下

Scottchou7, 2020-10-13:
解決了，謝謝您

closed 2020-10-13 by enazoe