enazoe/yolo-tensorrt #71: How to use the trt engine for the folder with images?

Justin-king-de, 2020-10-24:
@enazoe . First of all, thanks for sharing your wonderful work with us. I have a trained custom yolov4 model and I have created a trt engine file for that using your repo. Now in the` sample_detector.cpp` file, I see that the detections are being done for 2 images. However I want to run the inference process for many images that I have in a directory. Is there any function written for that in repo? Or any kind of idea or help in this regard will be much helpful to me. Thank you in advance.

enazoe, 2020-10-26:
just read the sample_detector.cpp ，use opencv read camera and save image

Justin-king-de, 2020-10-26:
@enazoe In my case, I don't have access to camera. I can only access images from a folder. In python it can be done easily. I was looking for a custom function in C++ which can do that.

Justin-king-de, 2020-10-28:
Hello @enazoe .I have somehow figured out on how to run it on multiple images from directory. On the detected images, its writing the id of the class using cv::imwrite. But I want to put the classname instead of its id. How to do that in sample_detector.cpp. I have tried. But was unable to do. Any suggestion will be of great help to me. Thank you in advance.

SinGit1, 2020-11-13:
@Justin-king-de can you share how you managed to run it on multiple images from the directory. I am not managing it at all.