enazoe/yolo-tensorrt #72: How to launch the example

xpeedster, 2020-10-26:
Hi, I am a little bit new into this. I generated the trt engine as explained in the README, but I would like to know how to execute the example provided.

xpeedster, 2020-10-26:
When I run ./yolo-trt it says:

File does not exist: ../configs/yolov4-kFLOAT-batch1.engine

xpeedster, 2020-10-26:
After building the engine, I run the yolo-trt and this error happens:

Loading TRT Engine...
ERROR: coreReadArchive.cpp (55) - Serialization Error in verifyHeader: 0 (Length in header does not match remaining archive length)
ERROR: INVALID_STATE: std::exception
ERROR: INVALID_CONFIG: Deserialize the cuda engine failed.
Loading Complete!
yolo-trt: /home/nvidia/yolo-tensorrt/modules/yolo.cpp:100: Yolo::Yolo(const NetworkInfo&, const InferParams&): Assertion `m_Engine != nullptr' failed.
Aborted (core dumped)

closed 2020-10-26 by xpeedster

freeluojing123, 2022-03-02:
Have you solved the problem yet? I've also encountered this problem

xpeedster, 2022-03-02:
No, I did not manage to solve it.