enazoe/yolo-tensorrt #80: Can you share the calibration Images File

Rhee-site, 2020-11-17:
Hi
First of all, thank you for sharing a great project.

There is no problem in compile.
but calibration image file is required when running test_dll.

Can you share the calibration Images file??

Thank you for your reply.

enazoe, 2020-11-17:
checkout the release tag [yolov5-2.0](https://github.com/enazoe/yolo-tensorrt/releases/tag/yolov5-2.0) about the calibration_images.rar

Rhee-site, 2020-11-17:
The data you sent and the required data are different.

The required data are as follows.

data/test/07/0.jpg
data/test/07/1.jpg
data/test/07/2.jpg
data/test/07/3.jpg
data/test/07/4.jpg
data/test/07/5.jpg
data/test/07/6.jpg
data/test/07/7.jpg
data/test/07/8.jpg
data/test/07/9.jpg
data/test/07/10.jpg
data/test/07/11.jpg
data/test/07/12.jpg
data/test/07/13.jpg
data/test/07/14.jpg
data/test/07/15.jpg
data/test/07/16.jpg
data/test/07/17.jpg
data/test/07/18.jpg
data/test/07/19.jpg
data/test/07/20.jpg
data/test/07/21.jpg
data/test/07/22.jpg
data/test/07/23.jpg
data/test/07/24.jpg
data/test/07/25.jpg
data/test/07/26.jpg
data/test/07/27.jpg
data/test/07/28.jpg
data/test/07/29.jpg
data/test/07/30.jpg
data/test/08/0.jpg
data/test/08/1.jpg
data/test/08/2.jpg
data/test/08/3.jpg
data/test/08/4.jpg
data/test/08/5.jpg
data/test/08/6.jpg
data/test/08/7.jpg
data/test/08/8.jpg
data/test/08/9.jpg
data/test/08/10.jpg
data/test/08/11.jpg
data/test/08/12.jpg
data/test/08/13.jpg
data/test/08/14.jpg
data/test/08/15.jpg
data/test/08/16.jpg
data/test/08/17.jpg
data/test/08/18.jpg
data/test/08/19.jpg
data/test/08/20.jpg
data/test/08/21.jpg
data/test/08/22.jpg
data/test/08/23.jpg
data/test/08/24.jpg
data/test/08/25.jpg
data/test/08/26.jpg
data/test/08/27.jpg
data/test/08/28.jpg
data/test/08/29.jpg
data/test/08/30.jpg
data/test/11/0.jpg
data/test/11/1.jpg
data/test/11/2.jpg
data/test/11/3.jpg
data/test/11/4.jpg
data/test/11/5.jpg
data/test/11/6.jpg
data/test/11/7.jpg
data/test/11/8.jpg
data/test/11/9.jpg
data/test/11/10.jpg
data/test/11/11.jpg
data/test/11/12.jpg
data/test/11/13.jpg
data/test/11/14.jpg
data/test/11/15.jpg
data/test/12/0.jpg
data/test/12/1.jpg
data/test/12/2.jpg
data/test/12/3.jpg
data/test/12/4.jpg
data/test/12/5.jpg
data/test/12/6.jpg
data/test/12/7.jpg
data/test/12/8.jpg
data/test/12/9.jpg
data/test/12/10.jpg
data/test/12/11.jpg
data/test/12/12.jpg
data/test/12/13.jpg
data/test/12/14.jpg
data/test/12/15.jpg
data/test/13/0.jpg
data/test/13/1.jpg
data/test/13/2.jpg
data/test/13/3.jpg
data/test/13/4.jpg
data/test/13/5.jpg
data/test/13/6.jpg
data/test/13/7.jpg
data/test/13/8.jpg
data/test/13/9.jpg
data/test/13/10.jpg
data/test/13/11.jpg
data/test/13/12.jpg
data/test/13/13.jpg
data/test/13/14.jpg
data/test/13/15.jpg

Any answers would be appreciated.

enazoe, 2020-11-17:
@Rhee-site pull the latest calibration_images.txt

Rhee-site, 2020-11-17:
Thank you for your reply.

It's working.

huunhan312, 2021-02-01:
@Rhee-site  I am sorry but I cannot find any images. Please tell me how to get the images for calibration.

Thank you !

enazoe, 2021-02-02:
@huunhan312 https://github.com/enazoe/yolo-tensorrt/releases/download/calibration_images.txt/calibration_images.rar

huunhan312, 2021-02-02:
> @huunhan312 https://github.com/enazoe/yolo-tensorrt/releases/download/calibration_images.txt/calibration_images.rar

Many thanks