enazoe/yolo-tensorrt #82: tensorrt推理yolov5-v2模型，无法预测不出结果

FanZhang91, 2020-11-18:


enazoe, 2020-11-18:
https://github.com/enazoe/yolo-tensorrt/blob/master/modules/trt_utils.h#L217
https://github.com/enazoe/yolo-tensorrt/blob/master/modules/trt_utils.h#L221
把“hardswish”改成“leaky”

FanZhang91, 2020-11-18:
@enazoe 把“hardswish”改成“leaky”之后问题得到解决，棒棒的

closed 2020-11-18 by FanZhang91

alicera, 2020-12-14:
请问有不用DLA 的方法去转INT8 吗？
我用一般方式去转 无法预测出结果

enazoe, 2020-12-14:
> 请问有不用DLA 的方法去转INT8 吗？
> 我用一般方式去转 无法预测出结果

int8和dla没关系，你没结果也和dla没关系

zzcqinag, 2020-12-14:
请问下，改成“leaky”之后还是出不来结果是什么原因呢？自己重新编译，engine重新生成的，python可以出结果
转之后的cfg：
[net]
width=640
height=640
channels=3
nc=1
depth_multiple=0.33
width_multiple=0.5
anchors=10, 13, 16, 30, 33, 23, 30, 61, 62, 45, 59, 119, 116, 90, 156, 198, 373, 326
是还有其他的地方要改吗？非常感谢

enazoe, 2020-12-14:
> 请问下，改成“leaky”之后还是出不来结果是什么原因呢？自己重新编译，engine重新生成的，python可以出结果
> 转之后的cfg：
> [net]
> width=640
> height=640
> channels=3
> nc=1
> depth_multiple=0.33
> width_multiple=0.5
> anchors=10, 13, 16, 30, 33, 23, 30, 61, 62, 45, 59, 119, 116, 90, 156, 198, 373, 326
> 是还有其他的地方要改吗？非常感谢

正常情况改完就没问题了，再不行我就不知道了，用我提供的2.0模型试吧

zzcqinag, 2020-12-14:
就是用的2.0模型，是不是网络输入的尺寸不对？ @enazoe

enazoe, 2020-12-14:
@zzcqinag 你看对不对啊，肯定要改成你自己的啊，再不行看看检测阈值，把阈值调下去，你直接在群里说吧，