enazoe/yolo-tensorrt #84: Is it support Scaled-YOLOv4 ?

Scottchou7, 2020-11-27:
Scaled-YOLOv4 has already released. Is it supported?

[Scaled-YOLOv4: Scaling Cross Stage Partial Network](https://arxiv.org/pdf/2011.08036.pdf)
[Scaled-YOLOv4](https://github.com/WongKinYiu/ScaledYOLOv4)
[yolov4-csp.cfg](https://github.com/WongKinYiu/ScaledYOLOv4/tree/yolov4-csp/models)

enazoe, 2020-11-27:
is working on it

Scottchou7, 2020-11-27:
OK, Thanks!

DatDoc, 2020-12-18:
@enazoe you're working on it ?

enazoe, 2020-12-19:
@Scottchou7 @DatDoc  yolov4-csp is ok now at branch [yolov4_scaled](https://github.com/enazoe/yolo-tensorrt/tree/yolov4_scaled)

DatDoc, 2020-12-19:
@enazoe Can it run on google colab ? Btw you're doing some great work, keep it up

auto-amc, 2021-06-04:
@enazoe! Please update code to use last yolov4-csp.cfg from https://github.com/AlexeyAB/darknet

LukeAI, 2021-06-20:
@enazoe  is yolor in the works?
https://github.com/WongKinYiu/yolor

auto-amc, 2021-06-21:
Hi!i am getting the following errorUsing torch 1.9.0 CUDA:0 (GeForce GTX 1070, 8116MB) Traceback (most recent call last):  File "test.py", line 330, in <module>    save_conf=opt.save_conf,  File "test.py", line 63, in test    model = Darknet(opt.cfg).to(device)  File "~/yolor/models/models.py", line 530, in __init__    self.module_defs = parse_model_cfg(cfg)  File "~/yolor/utils/parse_config.py", line 50, in parse_model_cfg    assert not any(u), "Unsupported fields %s in %s. See https://github.com/ultralytics/yolov3/issues/631" % (u, path)AssertionError: Unsupported fields ['objectness_smooth', 'resize', 'obj_normalizer', 'new_coords', 'max_delta'] in ~/darknet/cfg/yolov4-csp.cfg 20.06.2021, 20:43, "LukeAI" ***@***.******@***.*** is yolor in the works?https://github.com/WongKinYiu/yolor