enazoe/yolo-tensorrt #96: int8 is error

zhixiangzhangzstu, 2021-01-11:
hi,my int8 quantification is error

terminate called after throwing an instance of 'std::out_of_range'
  what():  _Map_base::at
Aborted (core dumped)

ZhenshengLee, 2021-02-20:
I have the same issue, any solution?

enazoe, 2021-02-20:
trim spaces at https://github.com/enazoe/yolo-tensorrt/blob/master/configs/yolov5-4.0/yolov5l.cfg#L3-L6 or remove the [blank line](https://github.com/enazoe/yolo-tensorrt/blob/master/configs/yolov5-4.0/yolov5l.cfg#L1)

Lenan22, 2022-10-31:
> hi,my int8 quantification is error
> 
> terminate called after throwing an instance of 'std::out_of_range' what(): _Map_base::at Aborted (core dumped)

Please refer to our open source quantization tool ppq, the quantization effect is better than the calibration of tensorrt, if you encounter issues, we can help you solve them.
https://github.com/openppl-public/ppq/blob/master/md_doc/deploy_trt_by_OnnxParser.md

Lenan22, 2022-10-31:
> I have the same issue, any solution?

Please refer to our open source quantization tool ppq, the quantization effect is better than the calibration of tensorrt, if you encounter issues, we can help you solve them.
https://github.com/openppl-public/ppq/blob/master/md_doc/deploy_trt_by_OnnxParser.md