enazoe/yolo-tensorrt #98: int8预测结果不对

gentlebreeze1, 2021-01-15:
yolov5s trt  float16可以准确的预测框，但是int8预测效果很差，该怎么解决呢 @enazoe

enazoe, 2021-01-16:
int8是会有精度损失，要用训练集的子集作为校准图片

PowerDi, 2021-03-18:
> int8是会有精度损失，要用训练集的子集作为校准图片

用ubuntu的TensorRT7.0.0 量化是没问题的。用Agx的TensorRT7.1.3(只有这个版本)，int8校准后结果不对了，fp32正常。代码上有什么需要注意的地方吗？

persuelx, 2022-10-23:
@PowerDi  请问后来你在Agx上这个int8的问题解决了吗？

PowerDi, 2022-10-23:
没有解决，可以转成onnx并使用端到端推理，不采用层api的形式

persuelx, 2022-10-24:
@PowerDi 好的，谢谢，那用onnx 端对端推理的话，可以解决上述那个问题了么？

Lenan22, 2022-10-31:
> yolov5s trt float16可以准确的预测框，但是int8预测效果很差，该怎么解决呢 @enazoe

Please refer to our open source quantization tool ppq, the quantization effect is better than the calibration of tensorrt, if you encounter issues, we can help you solve them.
https://github.com/openppl-public/ppq/blob/master/md_doc/deploy_trt_by_OnnxParser.md