enazoe/yolo-tensorrt #101: couldn't parse yolov4-tiny

mltloveyy, 2021-02-02:
Due to the different environment, I created a new project on VS2015 in windows10, and added your and tensorRT code to the project, the program was compiled and passed
When I tried to load a trained model based on yolov4-tiny, there was a memory out-of-bounds problem in line 1211 of trt_utils.cpp, and I tried to use yolov4-tiny and its pre-trained model yolov4-tiny.conv.29, the same problem occurred. All prompt errors in the 11th convolutional layer
https://github.com/enazoe/yolo-tensorrt/blob/322a1310adf28fb911f1e10415aa5de6432126aa/modules/trt_utils.cpp#L1211
`File does not exist : D:/Code/yolo-tensorrt/configs/yolov4-tiny-kHALF-batch1.engine
Loading pre-trained weights...
Invalid network type
Loading complete!
      layer               inp_size            out_size       weightPtr
(1)   conv-bn-leaky     3 x 416 x 416      32 x 208 x 208    992
(2)   conv-bn-leaky    32 x 208 x 208      64 x 104 x 104    19680
(3)   conv-bn-leaky    64 x 104 x 104      64 x 104 x 104    56800
(4)   chunk            64 x 104 x 104      32 x 104 x 104    56800
(5)   conv-bn-leaky    32 x 104 x 104      32 x 104 x 104    66144
(6)   conv-bn-leaky    32 x 104 x 104      32 x 104 x 104    75488
(7)   route                  -             64 x 104 x 104    75488
(8)   conv-bn-leaky    64 x 104 x 104      64 x 104 x 104    79840
(9)   route                  -            128 x 104 x 104    79840
(10)  maxpool         128 x 104 x 104     128 x  52 x  52    79840`

mltloveyy, 2021-02-02:
I tried yolov4, and the same error was reported but in the 12th convolutional layer

Nuzhny007, 2021-02-02:
Do you change net_type in config?

closed 2021-02-02 by mltloveyy