enazoe/yolo-tensorrt #102: How to get the count of no.of objects detected in a frame?

Justin-king-de, 2021-02-03:
Hey @enazoe I have used your repo to convert yolov4 darknet weights into tensorrt engine files. Now I want to get the count of no.of objects detected in a frame.How can I get that? Is there any way in the repo to count number of objects detected in a frame? Your help will be appreciated.

enazoe, 2021-02-05:
@Justin-king-de  https://github.com/enazoe/yolo-tensorrt/blob/master/modules/class_detector.h#L15  the size of this vector

Justin-king-de, 2021-02-05:
@Justin-king-de Thanks for reply.I got the value of number of detections from the line pointed by you. But I want to actually display that value on the detected image . How can I achieve that in `sample_detector.cpp` script. In that script, `detect` function is taking vector of type `BatchResult `as input but not of type `Result.` So how can I access that value from `sample_detector.cpp` and write it on my image with detections. Please help me in this thing.

IsraelLencina, 2021-02-09:
BatchResult is a vector of Result, so making a size should work. If you want to display it, should write text in with:

```
cv::putText(img, // target image
            **number_of_detections.to_string()**, // text
            cv::Point(10, img.rows / 2), // top-left position
            cv::FONT_HERSHEY_DUPLEX,
            1.0,
            CV_RGB(118, 185, 0), //f ont color
            2);
```

Justin-king-de, 2021-02-09:
@IsraelLencina Okay. Thanks.

closed 2021-02-19 by Justin-king-de