enazoe/yolo-tensorrt #115: 精度不一致问题

PowerDi, 2021-04-06:
想请教下有没有验证过精度差别问题？(fp32)模式

同样的图片在训练环境的测试环境中没有出现误检的情况，但是在AGX平台的tensorRT推理中，则出现了与训练环境不一样的误检（或者是突然很大的一个框），平台出现了Some tactics do not have sufficient workspace memory to run. Increaseing workspace size may increase performance的提示有关系吗？

enazoe, 2021-04-06:
没出现过，再就是engine文件不通用，在jetson平台需要重新生成engine

PowerDi, 2021-04-06:
> 没出现过，再就是engine文件不通用，在jetson平台需要重新生成engine

正常的物体在agx上置信度推理出来的高1%左右，但是会有误检

PowerDi, 2021-04-16:
您好，这边在学习的过程中遇到了一些新的问题，可以帮忙看一下吗？
我将网络的输入修改成了512尺寸，并将focus层之后的卷积层的步长改为了1想要让yolo层的特征尺度放大一倍，诸如
```
# YOLOv5 backbone
backbone:
  # [from, number, module, args]
  [[-1, 1, Focus, [64, 3]],  # 0-P1/2
   [-1, 1, Conv, [128, 3, 1]],  # 1-P2/4
```
但我将训练后的模型在trt转换后所有框的置信度输出，发现结果都是异常值，于是我进行了逐层分析。
发现在将focus层后的CBL层，同样的位置在不修改其他参数只修改步长由2->1之后，出现了训练环境pytorch与TRT输出不一样的结果（正常模型在该层输出是一样的）
于是我将该层的输出逐个导入到pytorch相应的层输出中看是否能推理成功，发现最后结果也是异常了，但同样的导出方法在步长没改1的模型中可以将值逐个导入到pytorch进行推理，可见focus层后的CBL层输出结果不对之外，也不是因为精度损失导致的，有解决的思路吗？即
```
nvinfer1::ILayer * out = layer_conv_bn_act(trtWeights,
				"model."+std::to_string(i-1), model_wts, previous, m_Network, n_out_channel, kernel_size, stride);
```
只将stride设成1.此时输出维度是对应128*256*256

zhangjian-neu, 2021-07-19:
是的。当使用yolov3-voc.cfg时，此程序的输出与darknet的输出不一致。简单测试以下 darknet 附带的 person.jpg就会发现。