enazoe/yolo-tensorrt #121: terminate called after throwing an instance of 'std::out_of_range'

henbucuoshanghai, 2021-06-02:
## Env

- GPU, RTX2060
- Ubuntu18.04,
- Cuda11.0
- TensorRT7.2.2

## About this repo

- which model? yolov5 

## Your problem
i use the yolo5s model,get the engine and test it all right..
sudo ./yolov5 -s yolov5s.wts yolov5s.engine s
sudo ./yolov5 -d yolov5s.engine ../samples
when i use fp32 and fp16,everyting is right!!!  

but int8 is wrong.
Loading weights: yolov5s.wts
Your platform support int8: true
Building engine, please wait for a while...
reading calib cache: int8calib.table
[06/02/2021-16:54:38] [W] [TRT] TensorRT was linked against cuDNN 8.0.5 but loaded cuDNN 8.0.4
terminate called after throwing an instance of 'std::out_of_range'
  what():  _Map_base::at
Aborted

what is wrong?

enazoe, 2021-06-02:
```
sudo ./yolov5 -s yolov5s.wts yolov5s.engine s
sudo ./yolov5 -d yolov5s.engine ../samples
```
this is not my project ....

henbucuoshanghai, 2021-06-04:
yes.....      in wrong project.

by the way ,the yolo 3 and yolo4 in this git are  trained  under the officifal darnket?

enazoe, 2021-06-04:
> yes..... in wrong project.
> 
> by the way ,the yolo 3 and yolo4 in this git are trained under the officifal darnket?

yes

henbucuoshanghai, 2021-06-04:
tks for your help,useful

but yolo34 is in c language,how to change it to tensorrt api?using onnx?or directly using the trt api?
where is the related code?

enazoe, 2021-06-05:
in my project , I use the trt api build the yolov3 yolov4 model,and parse the darknet weight directly.