enazoe/yolo-tensorrt #123: YOLOv5s6 NMS result

sammilei, 2021-06-26:
Hi,

Thanks for making this repo and easy API. 
I am trying out the YOLOv5 module with a customized YOLOv5s6 weights.
The engines (INT8, FP16, and FP32) can run well, but the NMS result doesn't seem correct. 
[image] 
[image]

I tried changing the nms thresh, but it doesn't help. Do you see result like this when testing the `diou_nms()` function in `trt_utils.cpp`. Please advise. 

Thank you!

positive666, 2021-12-22:
Use tensorrt version 8.2

yang-gis, 2022-01-17:
> Hi,
>
> Thanks for making this repo and easy API. I am trying out the YOLOv5 module with a customized YOLOv5s6 weights. The engines (INT8, FP16, and FP32) can run well, but the NMS result doesn't seem [...]

hello,have you solved this problem?I have met the same problem.I tried trt7.x、trt8.0 and trt8..2. it don't work!

sunchuanxi, 2022-02-18:
i have the same problem,not solve

Nuzhny007, 2022-02-18:
Do you can share data (cfg, weights and image or video)? I'll try to reproduce and debug this case