enazoe/yolo-tensorrt #136: build error : ptxas fatal   : Value 'sm_30' is not defined for option 'gpu-name'

pmankyu, 2021-08-05:
A problem occurred during build

The build msg is: 

`pmk@pmk-ubuntu:~/yolo-tensorrt/build$ cmake ..
-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- The CUDA compiler identification is unknown
-- Check for working CUDA compiler: /usr/bin/nvcc
-- Check for working CUDA compiler: /usr/bin/nvcc -- broken
CMake Error at /usr/share/cmake-3.10/Modules/CMakeTestCUDACompiler.cmake:46 (message):
  The CUDA compiler

    "/usr/bin/nvcc"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /home/pmk/yolo-tensorrt/build/CMakeFiles/CMakeTmp
    
    Run Build Command:"/usr/bin/make" "cmTC_14f93/fast"
    /usr/bin/make -f CMakeFiles/cmTC_14f93.dir/build.make CMakeFiles/cmTC_14f93.dir/build
    make[1]: 디렉터리 '/home/pmk/yolo-tensorrt/build/CMakeFiles/CMakeTmp' 들어감
    Building CUDA object CMakeFiles/cmTC_14f93.dir/main.cu.o
    /usr/bin/nvcc     -x cu -c /home/pmk/yolo-tensorrt/build/CMakeFiles/CMakeTmp/main.cu -o CMakeFiles/cmTC_14f93.dir/main.cu.o
    ptxas fatal   : Value 'sm_30' is not defined for option 'gpu-name'
    CMakeFiles/cmTC_14f93.dir/build.make:65: recipe for target 'CMakeFiles/cmTC_14f93.dir/main.cu.o' failed
    make[1]: *** [CMakeFiles/cmTC_14f93.dir/main.cu.o] Error 255
    make[1]: 디렉터리 '/home/pmk/yolo-tensorrt/build/CMakeFiles/CMakeTmp' 나감
    Makefile:126: recipe for target 'cmTC_14f93/fast' failed
    make: *** [cmTC_14f93/fast] Error 2
    

  

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:3 (enable_language)

-- Configuring incomplete, errors occurred!
`
[image]

enazoe, 2021-08-06:
@pmankyu set the nvcc path to your environment variables

pmankyu, 2021-08-06:
@enazoe I have already setting in bashrc

export PATH=/usr/local/cuda-11.4/bin:$PATH
export LD_LIBRARY_PATH=/usr/local/cuda-11.4/lib64:$LD_LIBRARY_PATH
export CUDA_HOME=/usr/local/cuda-11.4

Wilson-shuo, 2021-08-06:
I also meet this bug.
Here is the way how i sloved it.
Add extra additional instructions behind your cmake order like below：
`sudo cmake .. -DCMAKE_CUDA_COMPILER:PATH=/usr/local/cuda-10.2/bin/nvcc`
(Remember to change your own CUDA path)
That could appoint compiler of cudacxx.
After that, you can compile successfully

pmankyu, 2021-08-06:
@Wilson-shuo Thank you!
cmake problem is solved, but the following error occurs when "make"
[image]

Wilson-shuo, 2021-08-06:
It seemed that you are compiling this repo on a x86/x64 plantform, not on Jetson series.
I could not understand the Korean words behind the "Fatal Error".
I guess that means the cmake could not find nvinfer.h? This file belong to TensorRT sdk.
So Please make sure you have installed "Tensor RT SDK" correctly on your computer.

pmankyu, 2021-08-08:
@Wilson-shuo 
thank you
and my system is ubuntu 18.04, cuda 11.3, opencv 4.2.0, tensorrt 8.0.1, cudnn 8.2.1
As you said, I installed Tensorrt and when I "make" it fails with the error below.
What's the problem?
[image]

enazoe, 2021-08-08:
@pmankyu checkout the branch to TRT8

pmankyu, 2021-08-08:
@enazoe 
Thank you!
When I checked out and "make", the following error occurred
I also installed tensorRT normally, but I don't know what the problem is. 

[image]

pmankyu, 2021-08-08:
@enazoe 
I update latest cmake 3.21.1
and when I do 'cmake' at TRT8 branch

A warning message appears
[image]

pmankyu, 2021-08-08:
@enazoe 
I solve the problem fatal error NvInfer.h

change include path

#include "/usr/include/x86_64-linux-gnu/NvInfer.h" at /modules/plugin_factory.h

thank you all

closed 2021-08-08 by pmankyu