enazoe/yolo-tensorrt #139: Jetson pack 32.6.1 and TensorRT 8.0 support

adrianosantospb, 2021-08-17:
Hi,

I'm trying to build this repo using jetson-pack-32-6-1 and tensorrt 8.0, but I'm getting a stranger error (stop the process without apparent error) that it could have some relation of JP 32 and TRT 8. I'd like to know If someone here already tested on these versions. 

Tks.

enazoe, 2021-08-18:
try the branch TRT8

adrianosantospb, 2021-08-20:
Hi, @enazoe . Tks for your support. Well, I did it and now I'm getting a strange error. Look at:

yolo-tensorrt/modules/calibrator.cpp:26:
/usr/include/aarch64-linux-gnu/NvInfer.h:6669:18: error:   overriding 'virtual void nvinfer1::IInt8Calibrator::writeCalibrationCache(const void*, std::size_t) noexcept'
     virtual void writeCalibrationCache(const void* ptr, std::size_t length) noexcept = 0;
                  ^~~~~~~~~~~~~~~~~~~~~
CMakeFiles/detector.dir/build.make:62: recipe for target 'CMakeFiles/detector.dir/external/yolo-tensorrt/modules/calibrator.cpp.o' failed
make[2]: *** [CMakeFiles/detector.dir/external/yolo-tensorrt/modules/calibrator.cpp.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/detector.dir/all' failed
make[1]: *** [CMakeFiles/detector.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

Any idea?

ismet-dagli, 2021-09-08:
The problem is still there. Can someone update here after finding a solution/idea on what to do?