enazoe/yolo-tensorrt #140: fatal error: Nvinfer.h: No such file or directory

nFeus, 2021-08-18:
i'm trying to branch TRT8 with Jetson Xavier NX(jetpack 4.6)

Scanning dependencies of target detector
[  4%] Building CXX object CMakeFiles/detector.dir/modules/calibrator.cpp.o
In file included from /home/user/yolo-tensorrt/modules/trt_utils.h:47:0,
                 from /home/user/yolo-tensorrt/modules/ds_image.h:28,
                 from /home/user/yolo-tensorrt/modules/calibrator.h:29,
                 from /home/user/yolo-tensorrt/modules/calibrator.cpp:26:
/home/user/yolo-tensorrt/modules/plugin_factory.h:35:10: fatal error: Nvinfer.h: No such file or directory
 #include "Nvinfer.h"
                  ^~~~~~~~~~~
compilation terminated.

this error occurs when i build it.
how can i fix it?

adrianosantospb, 2021-08-20:
You must install the libnvinfer-plugin-dev :
sudo apt install libnvinfer-plugin-dev -y

nFeus, 2021-08-24:
#include "Nvinfer.h" -> #include "NvInfer.h"
                    TRT7                               TRT8

electronicYH, 2021-11-01:
@bkykent 
I tried as you said, but I meet a new problem, Could you please give me some advice?
The problem is show as below:
[image]
[image]

MHGL, 2021-11-05:
@bkykent it works, thank u