enazoe/yolo-tensorrt #146: Compiling Error TensorRT 6.5.0.1

hakan6710, 2021-09-28:
Hi,

first of all, thank you for this repo. It works very well on my Laptop.

I am trying to get it working on an NVIDIA Drive AGX with Drive OS 5.2.6.
It has TensorRT 6.5.0.1 for ARM64

I have a Problem with compiling the code:
`:2:
/home/nvidia/dev_ws_galactic/src/ros2-node/src/mot/detection/./yolo/modules/detect.h:121:15: error: looser throw specifier for ‘virtual const char* nvinfer1::DetectPluginCreator::getPluginNamespace() const’
   const char* getPluginNamespace() const override;
               ^~~~~~~~~~~~~~~~~~
In file included from /usr/include/aarch64-linux-gnu/NvInferRuntime.h:59:0,
                 from /usr/include/aarch64-linux-gnu/NvInfer.h:53,
                 from /home/nvidia/dev_ws_galactic/src/ros2-node/src/mot/detection/./yolo/modules/mish.h:6,
                 from /home/nvidia/dev_ws_galactic/src/ros2-node/src/mot/detection/./yolo/modules/trt_utils.h:38,
                 from /home/nvidia/dev_ws_galactic/src/ros2-node/src/mot/detection/./yolo/modules/ds_image.h:28,
                 from /home/nvidia/dev_ws_galactic/src/ros2-node/src/mot/detection/./yolo/modules/class_yolo_detector.hpp:5,
                 from /home/nvidia/dev_ws_galactic/src/ros2-node/src/mot/detection/class_detector.cpp:2:
/usr/include/aarch64-linux-gnu/NvInferRuntimeCommon.h:904:30: error:   overriding ‘virtual const AsciiChar* nvinfer1::IPluginCreator::getPluginNamespace() const noexcept’
     virtual AsciiChar const* getPluginNamespace() const noexcept = 0;
                              ^~~~~~~~~~~~~~~~~~
make[2]: *** [CMakeFiles/detector.dir/src/mot/detection/yolo/modules/calibrator.cpp.o] Error 1
make[2]: *** [CMakeFiles/detector.dir/src/mot/detection/yolo/modules/ds_image.cpp.o] Error 1
make[2]: *** [CMakeFiles/detector.dir/src/mot/detection/class_detector.cpp.o] Error 1
make[1]: *** [CMakeFiles/detector.dir/all] Error 2
make: *** [all] Error 2
`

enazoe, 2021-09-28:
does not support trt 6.x now

hakan6710, 2021-09-28:
Do you think that it would be easy to adapt to it ?

hakan6710, 2021-10-05:
I was able to solve the issue by using the changes from the TensorRT 8 branch. I am guessing that nvidia continues to update tensorRT for the drive platform and  6.5.3 is similiar to 8.X

RizM9292, 2022-10-16:
> I was able to solve the issue by using the changes from the TensorRT 8 branch. I am guessing that nvidia continues to update tensorRT for the drive platform and 6.5.3 is similiar to 8.X

Hi, 
Could you elaborate on what changes you made from TensorRT 8 branch for it to work with 6.5.3, please?

h-koese, 2022-10-17:
I am not sure anymore. I think I used the master branch and changed a few function prototypes. 

For what do you want to use this repo? Are you working this Drive OS and AGX?

RizM9292, 2022-10-17:
Hey,

I understand. I want to use it in Jetson Nano and Xavier with TRT 6 for my
custom model. Let me know if you can remember some of the changes.

Thank you!

hakan6710, 2022-10-17:
Do you have an Error Message ? Could you show me it ? @RizM9292