enazoe/yolo-tensorrt #156: problem about "override",

electronicYH, 2021-11-04:
I train the yolov5x release 6.0, with the project: ultralytics/yolov5
I meet the problem about " no Nvinfer.h"
I tried as bkykent said:
"#include "Nvinfer.h" -> #include "NvInfer.h"
"I" changed lower case to upper case"
, but I meet a new problem, Could you please give me some advice?

The problem is show as below:

[image]
[image]

enazoe, 2021-11-05:
the trt8 branch code need tensorrt 8.0

enazoe, 2022-05-05:
> hello~ I have the same environment on TX2 as you. How did you solve this problem?

update the jetpack,the trt8 branch code need tensorrt version >= 8.0