enazoe/yolo-tensorrt #157: INVALID_ARGUMENT: getPluginCreator could not find plugin CHUNK_TRT version 1.0

MHGL, 2021-11-09:
# 环境配置
- OS: Ubuntu20.04
- python3.7
- cuda11.0 + cudnn8.0
- tensorrt 7.1.3.4
- branch: master
- cfg: yolov4-tiny

# 问题描述
当使用tensorrt python api 加载engine时出现错误，提示CHUNK_TRT插件找不到
```
[TensorRT] ERROR: INVALID_ARGUMENT: getPluginCreator could not find plugin CHUNK_TRT version 1.0
[TensorRT] ERROR: safeDeserializationUtils.cpp (323) - Serialization Error in load: 0 (Cannot deserialize plugin since corresponding IPluginCreator not found in Plugin Registry)
[TensorRT] ERROR: INVALID_STATE: std::exception
[TensorRT] ERROR: INVALID_CONFIG: Deserialize the cuda engine failed.
Traceback (most recent call last):
  File "sample_detector.py", line 63, in <module>
    Detector(trt_file)()
  File "sample_detector.py", line 21, in __init__
    self.context = engine.create_execution_context()
AttributeError: 'NoneType' object has no attribute 'create_execution_context'
```

# 复现代码
```
TRT_LOGGER = trt.Logger(trt.Logger.VERBOSE)
# build engine
trt.init_libnvinfer_plugins(TRT_LOGGER, '')
PLUGIN_CREATOR = trt.get_plugin_registry().plugin_creator_list
with open(trt_file, 'rb') as f, trt.Runtime(TRT_LOGGER) as runtime:
      engine = runtime.deserialize_cuda_engine(f.read())
self.context = engine.create_execution_context()
```

enazoe, 2021-11-10:
engine 暂时没能用于python，因为里面有自定义插件，需要编译自定义插件的so

MHGL, 2021-11-10:
感谢您的回复。我尝试过编译插件so库（通过pytorch cuda extension 或者 CMakeLists中类似libdetector），这方面确实经验比较少，您可以给点建议吗？