enazoe/yolo-tensorrt #164: yolov4 inference speed is slow

Tristesse-stk, 2022-02-09:
Hi, thank you for sharing a great open source project about TensorRT.
There is a question after I run the sample_detector with yolov4 successfully.
The inference time I tested is longer than the time you mentioned in your benchmark.
My environment:
Jetson XAVIER NX
TensorRT7.1.3
cuda10.2
cudnn8.0.0
opencv4.1.1

My result table is below.
<html>
<body>

<span>

精度 | 图像尺寸 | 速度
-- | -- | --
FP32 | 416*416 | 180ms
FP16 | 416*416 | 90ms

</span>

</body>
</html>

It is seen that it is slower than yours.
[image]

If you can help me suggest some possible solutions, I will be very grateful and give you a star.

enazoe, 2022-02-09:
@Tristesse-stk modify the power mode to 2 core 12W, and turbo the cpu

Tristesse-stk, 2022-02-09:
Great！Your solution is helpful and the inference speed is as fasf as yours.I just gave the star to your project.

Tristesse-stk, 2022-02-11:
In the upper right corner of the interface.
[image]

Nuzhny007, 2022-02-11:
@denred0 but it works only for Nvidia Jetson

denred0, 2022-02-11:
@Nuzhny007 Yes, I understood

closed 2022-02-12 by enazoe