enazoe/yolo-tensorrt #171: TRT8版 crash

tuteming, 2022-03-10:
我用master版by TensorRT-7.2.1.6, 一切ok.

當用TRT8版by  TensorRT-8.0.1.6, compile ok, yolo-trt.exe and detector.dll 也都產生
run yolo-trt.exe,出現
File does not exist : ../configs/yolov4-kHALF-batch1.engine
Loading pre-trained weights...
Loading complete!
........
過一下yolo-trt.ex就跳掉了
請賜教,謝謝

enazoe, 2022-03-10:
@tuteming 看下trt和cuda cudnn的版本是不是匹配，这种问题一般情况下是环境问题

tuteming, 2022-03-10:
trt和cuda cudnn的版本是是匹配的

tuteming, 2022-03-10:
另一個問題,我用master版跑
偵測都對,but in command 視窗,出現的是無窮迴圈,同一張圖一直偵測,輸出訊息一直跑,
請問這是甚麼問題?也請賜教,謝謝
Loading TRT Engine...
Loading Complete!
pre elasped time:3.3799ms
inference elasped time:318.334ms
post elasped time:0.4608ms
detect elasped time:324.954ms
batch 0 id:4 prob:0.999942 rect:[114 x 131 from (621, 611)]
batch 0 id:4 prob:0.999902 rect:[209 x 63 from (199, 488)]
batch 0 id:4 prob:0.999899 rect:[250 x 81 from (185, 415)]
batch 0 id:4 prob:0.999696 rect:[210 x 67 from (211, 455)]
batch 0 id:4 prob:0.999689 rect:[32 x 106 from (97, 434)]
batch 0 id:4 prob:0.998361 rect:[109 x 137 from (605, 643)]
pre elasped time:3.3904ms
inference elasped time:20.962ms
post elasped time:0.4103ms
detect elasped time:26.4252ms
batch 0 id:4 prob:0.999942 rect:[114 x 131 from (621, 611)]
batch 0 id:4 prob:0.999902 rect:[209 x 63 from (199, 488)]
batch 0 id:4 prob:0.999899 rect:[250 x 81 from (185, 415)]
batch 0 id:4 prob:0.999696 rect:[210 x 67 from (211, 455)]
batch 0 id:4 prob:0.999689 rect:[32 x 106 from (97, 434)]
batch 0 id:4 prob:0.998361 rect:[109 x 137 from (605, 643)]
pre elasped time:4.7177ms
inference elasped time:21.1883ms
post elasped time:0.4125ms
detect elasped time:28.917ms
.......

enazoe, 2022-03-10:
程序里有for循环，看一眼代码就知道了

tuteming, 2022-03-10:
請問你指的for循环是哪裡?下面我mark loop,結果還是一樣.
                  int i = 0;// only one image
		//disp
		//for (int i = 0; i < batch_img.size(); ++i)
		//{
			for (const auto &r : batch_res[i])
			{
				std::cout << "batch " << i << " id:" << r.id << " prob:" << r.prob << " rect:" << r.rect << std::endl;
				cv::rectangle(batch_img[i], r.rect, cv::Scalar(255, 0, 0), 2);
				std::stringstream stream;
				stream << std::fixed << std::setprecision(2) << "id:" << r.id << "  score:" << r.prob;
				cv::putText(batch_img[i], stream.str(), cv::Point(r.rect.x, r.rect.y - 5), 0, 0.5, cv::Scalar(0, 0, 255), 2);
			}
			cv::namedWindow("image" + std::to_string(i), cv::WINDOW_NORMAL);
			cv::imshow("image" + std::to_string(i), batch_img[i]);
		//}

tuteming, 2022-03-10:
thanks, I known.

closed 2022-03-10 by enazoe