enazoe/yolo-tensorrt #173: YOLOV4 INT8 Calibration ERROR

nameIsTooSimple, 2022-03-14:
File does not exist : ../configs/yolov4-kINT8-batch2.engine
Building the TensorRT Engine...
Using cached calibration table to build the engine
Using cached calibration table to build the engine
WARNING: TensorRT was linked against cuBLAS/cuBLAS LT 11.5.1 but loaded cuBLAS/cuBLAS LT 11.4.2
WARNING: Detected invalid timing cache, setup a local cache instead
ERROR: 2: [reformatRunner.cpp::onShapeChangeNONCONST::104] Error Code 2: Internal Error (Assertion mCombinedScalesSize >= channelDst failed.)
yolo-trt: /work/app/yolo-tensorrt/modules/yolo.cpp:488: void Yolo::createYOLOEngine(nvinfer1::DataType, Int8EntropyCalibrator*): Assertion `m_Engine != nullptr' failed.

Device: RTX A4000
OS: Ubuntu 20.04
CUDA 11.3, CUDNN 8.2.1, TensorRT 8.0.1.6

wangxudong-cq, 2022-06-09:
I had the same issue with xavier on yolov3/4/5.
ENV:
Xlib:  extension "NV-GLX" missing on display "localhost:11.0".
Xlib:  extension "NV-GLX" missing on display "localhost:11.0".
Xlib:  extension "NV-GLX" missing on display "localhost:11.0".
 - NVIDIA Jetson AGX Xavier [16GB]
   * Jetpack 4.6 [L4T 32.6.1]
   * NV Power Mode: MODE_15W_DESKTOP - Type: 7
   * jetson_stats.service: active
 - Libraries:
   * CUDA: 10.2.89
   * cuDNN: 8.2.1.32
   * TensorRT: 8.0.1.6
   * Visionworks: 1.6.0.501
   * OpenCV: 3.4.3 compiled CUDA: YES
   * VPI: ii libnvvpi1 1.1.15 arm64 NVIDIA Vision Programming Interface library
   * Vulkan: 1.2.70

wangxudong-cq, 2022-08-12:
the same issue, pls make a reply.

Lenan22, 2022-10-31:
> File does not exist : ../configs/yolov4-kINT8-batch2.engine Building the TensorRT Engine... Using cached calibration table to build the engine Using cached calibration table to build the engine [...]

Please refer to our open source quantization tool ppq, the quantization effect is better than the calibration of tensorrt, if you encounter issues, we can help you solve them.
https://github.com/openppl-public/ppq/blob/master/md_doc/deploy_trt_by_OnnxParser.md