enazoe/yolo-tensorrt #188: trt8 maxpool的問題

tuteming, 2023-02-17:
我前面的問題我發現是在trt8 的yolo.cpp-line 410 被comment掉, trt7則沒有comment
if (m_configBlocks.at(i).at("size") == "2" && m_configBlocks.at(i).at("stride") == "1")
            {
              //  m_TinyMaxpoolPaddingFormula->addSamePaddingLayer("maxpool_" + std::to_string(i));
            }
而我用了
[maxpool]
size=2
stride=2

在trt7成功偵測
在trt8則失敗 ( trt8 remove "setPoolingOutputDimensionsFormula")
請問可以解決這個問題嗎?

謝謝你花寶貴的時間看我的問題.

tuteming, 2023-02-17:
抱歉 我覺得問題也不在這  我再查查看 不過真是沒道理 trt7 ok trt8則失敗

tuteming, 2023-02-18:
sorry, 我應該確定是maxpool的問題,因在建enging時(以下我只列部分)
(11)  conv-bn-leaky    64 x 200 x 200     128 x 200 x 200    280160
(12)  skip            128 x 200 x 200     128 x 200 x 200        -
(13)  maxpool         128 x 200 x 200     128 x 100 x 100    280160
(14)  conv-bn-leaky   128 x 100 x 100     128 x 100 x 100    428128

WARNING: Weights [name=conv_11 + batch_norm_11.weight] had the following issues when converted to FP16:
WARNING:  - Subnormal FP16 values detected.
WARNING: If this is not the desired behavior, please modify the weights or retrain with regularization to reduce the magnitude of the weights.
WARNING: Weights [name=conv_14 + batch_norm_14 + PWN(leaky_14).weight] had the following issues when converted to FP16:
WARNING:  - Subnormal FP16 values detected.
WARNING: If this is not the desired behavior, please modify the weights or retrain with regularization to reduce the magnitude of the weights.

列config時沒錯 建enging時沒有13(maxpool)
請問這要改哪?  謝謝

lq0104, 2023-06-09:
这个升级到tensorrt8后NvInfer.h中的这个class TRT_DEPRECATED IOutputDimensionsFormula被遗弃了，所以m_TinyMaxpoolPaddingFormula也被注释掉了，我看这个代码的readme里也把对yolov tiny的支持去掉了，不知道后续用什么替代方案来支持呢？