enazoe/yolo-tensorrt #193: yolo7 tiny?

tugbakara, 2023-05-25:
Is this package usable for yolo7 tiny?

Nuzhny007, 2023-05-26:
Really, no. 
This project used a darknet-style config and it meed to manually support for all versions of new YOLOs.
I think that a simplier way - export a pt model to onnx, converting it to TensorRT and using.
In my project I'm going to this way.

tugbakara, 2023-05-26:
Actually  I am a bit away from this, looking around easy to use things, so in that case could you explain or give a link for the part that you said " - export a pt model to onnx, converting it to TensorRT" I need to learn how can I do that ?
Thanks!

Nuzhny007, 2023-05-27:
You can try this complete example: https://github.com/Linaom1214/TensorRT-For-YOLO-Series/tree/main

I'm using the same approach in my project: https://github.com/Smorodov/Multitarget-tracker/blob/master/src/Detector/tensorrt_yolo/YoloONNX.cpp
But it is the part of a big project.